i-am-shodan/USBArmyKnife #13: Connection error + unrecognised SD Card

LaiKash, 2024-10-14:
Trying to run the project as per the README I always get a connection error when accessing the web server:
[image]

Additionally no matter what I always get "Could not find SD card". I have the same sandsik as in the examples but I tried with other ones formatted in FAT32 and it's not recognised.

i-am-shodan, 2024-10-14:
That webpage is shown when the request to /data.json failed. It doesn't currently retry - it probably should. As a result sometimes I need to reload that page because the little webserver on the device isn't perfect.

What happens if you go to http://4.3.2.1/data.json with your browser?

i-am-shodan, 2024-10-14:
Ah I didn't notice the "Could not find SD card". The device is in an error state and is displaying that message before it restarts in 60s. No wonder the web interface didn't work! I'll make sure to turn off the WiFi etc when I detect this.

The hardware is a bit picky about the SD card you use. I had to go through a few different ones to find one that was compatible. Mine is formatted FAT32.

LaiKash, 2024-10-14:
> That webpage is shown when the request to /data.json failed. It doesn't currently retry - it probably should. As a result sometimes I need to reload that page because the little webserver on the device isn't perfect.
> 
> What happens if you go to http://4.3.2.1/data.json with your browser?

I get a 404 not found. I'm starting to think that the dongle is broken, it is impossible to detect the SD Card. No issue during the compilation phase, just some warnings. Tried with MacOS and Windows (both for compiling/uploading and to plug it in), same result

i-am-shodan, 2024-10-14:
I am near 100% sure its the SD card. The one you're using will never work.

LaiKash, 2024-10-14:
> I am near 100% sure its the SD card. The one you're using will never work.

Okey, I'll try some more, already did with 2: one Amazon Basics and a Sandisk 16GB. Any format options I should specify?

i-am-shodan, 2024-10-14:
I'm pretty sure I just did a normal FAT32 format in Windows.
I have a lot of SD cards that didn't work, i'll try changing some option and see if I can get any of them going.

i-am-shodan, 2024-10-14:
I've now got every one of my SDs to work by changing some of the formatting options. My plan for the next release is to allow you to format the SD card on the device. This might fix it for you.

LaiKash, 2024-10-14:
> I've now got every one of my SDs to work by changing some of the formatting options. My plan for the next release is to allow you to format the SD card on the device. This might fix it for you.

Thank you very much! Super appreciated!

eazeysec, 2024-10-14:
Hey I had this same probelm with a Amazon basics 128Gb and 256Gb.
Got round it by segmenting the first 32Gb of the SD card and then formatted that as FAT32.

I'm lazy so here are some AI quick instructions that work:

Step 1: Open Disk Management

    Press Windows Key + X and select Disk Management from the list.

    In the Disk Management window, locate your USB drive. Ensure you are selecting the correct drive (in this case, the drive letter is E:).

        Warning: Make sure you select the correct drive to avoid accidentally formatting another drive.

Step 2: Delete Existing Partition

    Right-click on the partition of the USB drive (shown as a volume like E:).
    Select Delete Volume. This will remove any existing partitions.

Step 3: Create a New Partition (32 GB)

    Right-click on the unallocated space of the USB drive.
    Select New Simple Volume.
    In the New Simple Volume Wizard, specify the volume size as 32768 MB (which is equal to 32 GB).
    Assign the drive letter E: to the partition.
    Choose FAT32 as the file system and proceed with formatting the partition.

LaiKash, 2024-10-15:
> Hey I had this same probelm with a Amazon basics 128Gb and 256Gb. Got round it by segmenting the first 32Gb of the SD card and then formatted that as FAT32.
>
> I'm lazy so here are some AI quick [...]

I think this is not just the issue. I have a 16GB SD Card which should drop under the FAT32 limit, it still doesn't recognise it. With the 512GB Amazon Basics one I have, same issue. 

I'm formatting it from a MacOS but it should be the same (I tried yesterday with Windows too):

```
# diskutil partitionDisk /dev/disk10 MBR FAT32 PART1 32G FAT32 PART2 R
/dev/disk10 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *512.1 GB   disk10
   1:                 DOS_FAT_32 PART1                   32.0 GB    disk10s1
   2:                 DOS_FAT_32 PART2                   480.1 GB   disk10s2
 ```
 
 Thanks for the insight!

i-am-shodan, 2024-10-15:
I've added some code that will enable you to format the SD card if it can't load it. Can you pull the latest code and try that.

If that doesn't work then it might well be a hardware issue.

LaiKash, 2024-10-15:
Still didn't work with 3 cards. Can be a HW issue, I'll contact the manufacturer. Thanks!

i-am-shodan, 2024-10-15:
Did the device come up with a screen asking you to format the card? If you formatted did it finish really fast?

LaiKash, 2024-10-15:
> Did the device come up with a screen asking you to format the card? If you formatted did it finish really fast?

Yes, it actually finished super fast, then it "rebooted". In fact I believe the format failed. I had the cards formatted in `Apple_HFS` and after formatting it kept it like that

i-am-shodan, 2024-10-15:
Yep your assumption is correct. It should take about 5m to format a card so returning instantly means from the hardware's point of view it can't be found.

Going to go with faulty hw.

I'm working on porting the code to a devices without a screen, led, SD etc. You will probably be able to use it with that so don't bin it.

i-am-shodan, 2024-10-16:
I've closed the issue as I think its hw related. I'm about to push a change to add support for devices without an SD card, if you don't mind hacking about potentially you could use some of the logic on your device.

closed 2024-10-16 by i-am-shodan

LaiKash, 2024-10-29:
Just updating that I confirmed it was a HW issue. Aliexpress stuff... Now I need to get the agent running and learn how to build new scripts! 

For example, I want to send the PCAP via the agent to a listening WireShark instead of just extracting the PCAP file. 

Something like this trick I use when I don't have a graphical interface but I do have SSH:

`ssh <IP> 'tcpdump -i <iface> -s0 -c 1000 -nn -w -' | wireshark -K -i -`

Thanks for the support @i-am-shodan and consider opening a `buymeacoffee`. Anything you need from mobile reveng, send me a ping at anytime!