i-am-shodan/USBArmyKnife #14: Can't install it on the stick

Daniel-Ventura81, 2024-10-15:
Hy,

i've tried several times, did all the stuff that is described here but always got stuck on some error with espmarauder.
Is there any other way to deploy the lilygo (with display) stick?

big thx for your help in advance!

i-am-shodan, 2024-10-15:
Perhaps you could elaborate on 'some error with espmarauder'

i-am-shodan, 2024-10-15:
[image]

You seeing these files?

Daniel-Ventura81, 2024-10-15:
thx for your reply! i have the ESP32Marauder folder but there is nothing in it. downloaded all that git stuff new but still nothing in it.

Daniel-Ventura81, 2024-10-15:
worked now, i added this folder manually and now it deployed correct. thx for the hint! no idea why this folder isn't included when cloning the git.

i-am-shodan, 2024-10-15:
You didn't run the `git submodule update --init` command so it didn't pull down those extra files. Ideally you should delete those files and run that command - in the future someone might change something in that dir.

Anyway glad it's working, closing as complete.

closed 2024-10-15 by i-am-shodan

i-am-shodan, 2024-10-15:
Did you 'git checkout' or download it from GitHub as a ZIP file?

i-am-shodan, 2024-10-15:
```
PS D:\temp> git clone https://github.com/i-am-shodan/USBArmyKnife
Cloning into 'USBArmyKnife'...
remote: Enumerating objects: 539, done.
remote: Counting objects: 100% (539/539), done.
remote: Compressing objects: 100% (376/376), done.
remote: Total 539 (delta 192), reused 422 (delta 129), pack-reused 0 (from 0)
Receiving objects: 100% (539/539), 10.59 MiB | 38.32 MiB/s, done.
Resolving deltas: 100% (192/192), done.
```
```
PS D:\temp> cd .\USBArmyKnife\
```
```
PS D:\temp\USBArmyKnife> git submodule update --init
Submodule 'lib/ESP32Marauder' (https://github.com/i-am-shodan/ESP32Marauder) registered for path 'lib/ESP32Marauder'
Cloning into 'D:/temp/USBArmyKnife/lib/ESP32Marauder'...
Submodule path 'lib/ESP32Marauder': checked out 'e3b40947f5dd4b95fa034680753dd22c39033742'
```