i-am-shodan/USBArmyKnife #19: Agent setup instructions need update?

cyberpunked1985, 2024-10-25:
Trying to setup Agent.
Installed .NET sdk-8.0.403-windows-x64-installer on my Windows 11 box.
Noticed README instructions point to "cd tools\SerialAgent" but im seeing only tools\Agent directory.
When trying to run dotnet publish -r win-x64 within tools\Agent directory I am getting all kinds of errors, not sure if I am missing a step.

cyberpunked1985, 2024-10-25:
Sorry, looks like I figured it out as soon as I posted this.
Running the following fixed the errors.
"dotnet nuget add source https://api.nuget.org/v3/index.json -n nuget.org"

cyberpunked1985, 2024-10-26:
Now I have the Agent up and "running" on a test machine according to the webgui however, when trying to execute a command from the webgui command tab here is the output from the logs tab. 

Running cmdline shutdown /s
Executing cmdline: shutdown /s
Reading cmd line 0
Error executing command

USB mode shows Serial+HID.  Why are my commands not working?

i-am-shodan, 2024-10-26:
I've updated the readme to point to the right directory.

You're getting that error because you are running the wrong command. That box you typed into is for running commands in the USB Army Knife scripting language.

You need 'AGENT_RUN shutdown /s'

As per the example here:

https://github.com/i-am-shodan/USBArmyKnife/blob/205b68a3242f3deb9fd21a0a298a840f776a07c8/examples/install_agent_and_run_command/us-autorun.ds#L45

cyberpunked1985, 2024-10-26:
> I've updated the readme to point to the right directory.
>
> You're getting that error because you are running the wrong command. That box you typed into is for running commands in the USB Army Knife [...]

Now that all makes sense! Thank you!

i-am-shodan, 2024-10-28:
I've updated the web interface so there is now a new box for running commands via the agent. Hopefully this should make things clearer

closed 2024-10-28 by i-am-shodan