i-am-shodan/USBArmyKnife #23: FEATURE SUGGESTION: Bridge USB NCM to wifi

GuizzyQC, 2024-11-05:
Being a microcontroller rather than a fully featured SBC there's tools that cannot be run on the ESP32. I'm thinking for instance of Responder, Inveigh or the Impacket toolset. If it were possible to bridge the NCM to Wifi, it would possible to force a computer to interact with such tools on a separate device (SBC, phone or computer) on an out-of-band network, avoiding the firewall and its IDS.

i-am-shodan, 2024-11-05:
Yes

The ESP API should allow you to bridge. You might also be able to set it to use the WiFi interface instead of creating a second one which would be simpler.

GuizzyQC, 2024-11-05:
I'm still waiting on my Lilygo to arrive, but I'll be trying to do that once I get it. I think much of the work is already done in this example: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/device/tusb_ncm/main/tusb_ncm_main.c

i-am-shodan, 2024-11-05:
The problem with that example is that it's pure idf and doesn't make use of the Arduino wrapping code that's used for WiFi/webserver.

I'd start by replacing the functions in usb_ncm_init() https://github.com/i-am-shodan/USBArmyKnife/blob/a9fb999b59971c4a90f8e73caa74576797e129f7/src/Devices/USB/USBNCM.cpp#L68

And instead call esp_wifi_internal_tx and esp_wifi_internal_free_rx_buffer

The problem is you need to be able to get packets out of the WiFi adapter and return them to the USB stack.

One of the reasons to move to arduino3 was to get better networking access but I haven't checked in on that.

i-am-shodan, 2024-11-05:
Oh nice, it looks like just calling esp_wifi_internal_reg_rxcb(ESP_IF_WIFI_STA, pkt_wifi2usb) will get your function called with the packet.

You'd need to do some filtering on the MAC address I suppose to only move the packets you want

i-am-shodan, 2024-11-05:
I think if you did that all the packets received on the USB end would end up on the WiFi interface and in the air. However to get these to your device you might have to monitor and decrypt the WiFi packets with Wireshark, they won't be destined for your device unless you implement a basic routing table.

primarysound, 2024-11-17:
I would like to be able to control Marauder via the uart console. In the T-Dongle, the uart is output by a separate output.

i-am-shodan, 2024-11-17:
> I would like to be able to control Marauder via the uart console. In the T-Dongle, the uart is output by a separate output.

@primarysound Can you create a new issue. This isn't related to USB NCM.

GuizzyQC, 2024-11-20:
I've been checking this for a while and lwip and netif used by the NCM is able to do bridging between interfaces (esp_netif_br_glue), but I have yet to find a way to make the Arduino WiFi library talk with it, I'm not sure they can even be made to cooperate. It seems to me that switching the wifi to also use lwip / netif would make bridging much simpler, but of course the web server side would also have be adjusted to match.

i-am-shodan, 2024-11-20:
What problems are you encountering?

Under the hood the Arduino platform code for ESP32 is just calling the underlying ESP-LWIP codebase. I'd love to know where the complexity is.

Worse case scenario you should be able to copy WiFi.cpp and make your changes there. ESPAsycWebserver shouldn't care either it's using the sockets layer right?

Can't you just use one of the internal ESP32 functions to get the WiFi netif and go bridge from there?

GuizzyQC, 2024-11-21:
I believe I might have the code done for the bridge. In lib\usb-ncm.c:

```
#include "esp_netif_br_glue.h"
```

```
    esp_netif_inherent_config_t esp_netif_br_config = {
        .flags = (esp_netif_flags_t)(ESP_NETIF_DHCP_SERVER | ESP_NETIF_FLAG_EVENT_IP_MODIFIED | ESP_NETIF_FLAG_AUTOUP),
        .ip_info = &ip_info,
        .get_ip_event = IP_EVENT_ETH_GOT_IP,
        .lost_ip_event = IP_EVENT_ETH_LOST_IP,
        .if_key = "bridge",
        .if_desc = "usb bridge device",
        .route_prio = 10};

    esp_netif_config_t netif_br_cfg = {
        .base = &esp_netif_br_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_BR,
    };
    // Bridge configuration
    bridgeif_config_t bridgeif_config = {
        .max_fdb_dyn_entries = 10, // maximum number of address entries in dynamic forwarding database
        .max_fdb_sta_entries = 2,  // maximum number of address entries in static forwarding database
        .max_ports = 2      // maximum number of ports the bridge can consist of
    };
    esp_netif_br_config.bridge_info = &bridgeif_config;

    memcpy(esp_netif_br_config.mac, lwip_addr, 6);
    esp_netif_t *br_netif = esp_netif_new(&netif_br_cfg);

    // Create new esp netif bridge glue instance
    esp_netif_br_glue_handle_t netif_br_glue = esp_netif_br_glue_new();
    esp_netif_br_glue_add_port(netif_br_glue, s_netif);
    esp_netif_t* wifi=NULL;
    wifi = esp_netif_get_handle_from_ifkey("WIFI_AP_DEF");
    esp_netif_br_glue_add_port(netif_br_glue, wifi);

    esp_netif_attach(br_netif, netif_br_glue);

    esp_netif_action_start(br_netif, 0, 0, 0);
```

But I find that this does not pass the linker due to undefined references, and being unfamiliar with embedded programming, am at my wits end trying to resolve.

i-am-shodan, 2024-11-21:
You'll need to start by posting what it complains about. If it's a linker error we might need to copy some code out of esp-idf the other framework for esp32s.

GuizzyQC, 2024-11-21:
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/LILYGO-T-Dongle-S3/src/lib/usb-ncm/usb_ncm.c.o:(.literal.usb_ncm_init+0x2c): undefined reference to `_g_esp_netif_netstack_default_br'
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/LILYGO-T-Dongle-S3/src/lib/usb-ncm/usb_ncm.c.o:(.literal.usb_ncm_init+0x44): undefined reference to `esp_netif_br_glue_new'
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/LILYGO-T-Dongle-S3/src/lib/usb-ncm/usb_ncm.c.o:(.literal.usb_ncm_init+0x48): undefined reference to `esp_netif_br_glue_add_port'    
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/LILYGO-T-Dongle-S3/src/lib/usb-ncm/usb_ncm.c.o: in function `usb_ncm_init':
C:\Users\userfolder\Nextcloud\Projects\USBArmyKnife/lib/usb-ncm/usb_ncm.c:192: undefined reference to `esp_netif_br_glue_new'
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \USBArmyKnife/lib/usb-ncm/usb_ncm.c:196: undefined reference to `esp_netif_br_glue_add_port'      
c:/users/userfolder/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \USBArmyKnife/lib/usb-ncm/usb_ncm.c:199: undefined reference to `esp_netif_br_glue_add_port'      
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LILYGO-T-Dongle-S3\firmware.elf] Error 1

i-am-shodan, 2024-11-21:
What happens if you add a CONFIG_ESP_NETIF_BRIDGE_EN to the platform in.ini. Looks like a lot of functionality is wrapped with that.

GuizzyQC, 2024-11-21:
I still get the same error in the linker