i-am-shodan/USBArmyKnife #26: Issue with AP Connection

CreativeWolf, 2024-11-23:
Hi,

Flashed the latest version available on the LilyGo T-Dongle S3.

Connection to Access Point iPhone14 keeps dropping in Android Phone - That seems like an Android Issue and with no solution.

Tried connecting from the same laptop where the T-Dongle S3 is connected to and yet, the status shows "Connection Error" and uptime shows "Unknown" tried refreshing the page.

Devtool shows, unable to fetch data.json

Appreciate any help

Thanks

i-am-shodan, 2024-11-23:
I suspect your issues are one in the same. If data.json can't be retrieved then something terrible is happening internally - possibly hardware related.

What happens when you run something like `curl 4.3.2.1:8000/data.json'. I suspect that it just times out.

You should try using a different SD card with nothing on it.

The code that creates data.json is linked below. You could comment out some of the lines starting with root and see which one is break.

https://github.com/i-am-shodan/USBArmyKnife/blob/faadcd0b70fd972918d236cdf46465c1b9a14b07/src/Comms/Web/WebServer.cpp#L144

There is a possibility that an upstream change has broken Arduino JSON handling. I'll flash a recent version to my device later and see if I can repro.

i-am-shodan, 2024-11-23:
Probably a bug. The device shouldn't have even loaded the web interface so I'm not sure what's wrong there.

i-am-shodan, 2024-11-23:
Happy to close this?

CreativeWolf, 2024-11-24:
Yes, please. Thank you. Closing the issue.

closed 2024-11-24 by CreativeWolf