i-am-shodan/USBArmyKnife #32: Device not entering HID mode most of the time

dzikuzplastiku, 2025-01-05:
Hi, I recently got my LilyGo T-Dongle S3, flashed the USBArmyKnife and everything was fine until I replugged the device. Then it started showing up as:
[image]
instead of this, the expected outcome:
[image]
I am not pressing the button while plugging, the screen shows:
Device now running
USB Mode: Serial
USB Class: HID

There are also no errors in the logs, website functions normally, commands are also executed without any errors, its just that nothing happens on a PC.
Sometimes when I plug the device its working properly and showing as a HID device, but its like once every 20 tries.

i-am-shodan, 2025-01-05:
My guess is that it's something to do with the WinUSB driver being loaded instead of the correct one AKA it's your machine not the device. WinUSB is a generic driver and with this in the way the keyboard driver isn't getting correctly loaded.

Things like the Arduino IDE/other flashing tools can sometimes cause this to be installed. Also trying to get the device to be mounted in WSL2 replaces the driver entry. Maybe you did this years ago and the USB Army Knife just happens to be using the same VID&PID.

Things to try:
* Download and run the Device Cleanup Tool, remove everything then plug the device in
* Research the usbip detach and unbind commands
* Change the VID/PID in the setting tab on the web interface

dzikuzplastiku, 2025-01-05:
changing com port somehow did the job for me, thanks for your help, closing

closed 2025-01-05 by dzikuzplastiku