i-am-shodan/USBArmyKnife #35: badusb attacks do not work and also firmware does not see agent.img

gliderexpert, 2025-01-08:
Hello! I have badusb attacking, vnc screen viewer, and also when loading agent.img (built according to the instructions) on the main page of the site in the agent section says not connected (I attach a screenshot and agent.img ).
Help please. thanks in advance!

Screen: 
[image]

agent.img: https://mega.nz/file/KJhlGDSL#HRBjydvZrVY9toT1kUXe9Iocj04Ve1YHW3GQkqECbPo

i-am-shodan, 2025-01-08:
Assuming English isn't your first language as your issue doesn't make a lot of sense. What is clear is that:
* The agent is not running on your victims machine. Either because:
  * It hasn't been installed correctly
  * It isn't running
* The agent cannot communicate to the device
  * Something else is using the COM port
  * It is configured to communicate with the wrong vid/pid (presumably because it has been changed in the settings)

Maybe you can answer these question with a yes/no:
* Is BadUSB working? Have you gotten the device to make key presses?
* What language & keyboard layout is the victim machine using?
* Have you run this attack? https://github.com/i-am-shodan/USBArmyKnife/tree/master/examples/install_agent_and_run_command
* Is dllhost.exe running on your victim machine?
* On your victim machine, open explorer and type %APPDATA% then press enter. Do you see a file called PortableApp.dll
* Can you see a COM port in device manager that if for your device?

gliderexpert, 2025-01-08:
1. with usb-army knife - no. 
2. english ( us )
3. yep
4. yes
5. yes
6. yes

i-am-shodan, 2025-01-08:
So you are telling me:
* You've run the 'install agent bad USB attack'
* The PortableApp.dll is in %APPDATA%
* dllhost.exe is running on your machine
* you have a COM port

BUT you haven't got the device to emit keypresses - that makes no sense. Can you tell me how the install script installed the agent?

gliderexpert, 2025-01-08:
I tried building the agent myself (strictly following the instructions in the README), but it didn’t work. Then my friend sent me his ready-made agent.img (it works perfectly for him), but it didn’t work for me either :(

i-am-shodan, 2025-01-08:
You can always download the latest agent from https://github.com/i-am-shodan/USBArmyKnife/actions/workflows/dotnet.yml?query=branch%3Amaster+is%3Asuccess

gliderexpert, 2025-01-08:
I used the prebuilt agent binaries, assembled them into agent.img in WSL, but still, nothing worked.

gliderexpert, 2025-01-08:
I don’t know how PortableApp.dll ended up in APPDATA/Roaming, but at least it’s there :)

i-am-shodan, 2025-01-08:
From the web interface go to the commands tab.

Expand Raw
[image]

Enter the command GUI and press Execute

Tell me what happens on your unlocked victim machine

gliderexpert, 2025-01-08:
nothing 🤷

gliderexpert, 2025-01-08:
Running cmdline GUI
Executing cmdline: GUI
Reading cmd line 0

i-am-shodan, 2025-01-08:
In device manager, when you plug in the device so you see a new 'HID Keyboard Device' appear under keyboards?

i-am-shodan, 2025-01-08:
OK there is something odd with your Windows box. It hasn't loaded the HID driver for the device. My guess is that you've got some other dev tool on your box that has replace the driver. I've seen issues in the past with things like the Arduino UI and even 3D printing tools like Cura.

My guess is that you plugged it in, it correctly used BadUSB to install the agent, the agent is now correctly running but it can't talk to the device because something else has stolen the COM port. On top of that some broken configuration of your machine is stopping the standard HID driver from loading.

Instructions that might help are here. https://github.com/i-am-shodan/USBArmyKnife/issues/32

gliderexpert, 2025-01-08:
I changed the PID and VID, and the HID functionality started working. But the agent is still in a "Not Connected" state, and VNC is not working.

i-am-shodan, 2025-01-08:
OK you definitely have some driver issue on your machine!

The reason why the agent is not working is because it was installed to talk to a specific VID and PID. To fix it you'll need to remove the agent and reinstall it. I have no instructions for that but briefly you'll need to:
* Stop and then delete the scheduled task created by the agent 'Security Script'
* Kill the dllhost.exe process
* Delete the agent from %APPDATA%
* Delete the agentInstalled file on your SD card

then reinstall the agent

gliderexpert, 2025-01-08:
Did everything you said, and it still doesn’t work :( HID is working, but the agent is still "Not Connected".

i-am-shodan, 2025-01-08:
Has the agent been reinstalled successfully?

1. If you unplug your device does the COM port go away? 
1. On your victim machine, open explorer and type %APPDATA% then press enter. Do you see a file called PortableApp.dll
1. Is dllhost.exe running on your victim machine?
1. If you connect to the COM port with Putty do you get an error? An error is good.
[image]
[image]
1. If you look in the scheduled tasks do you see Security Script? Does it correctly display the VID and PID you expect ie changed your device to? This value is in hex.
[image]
1. Are you running any programs that might be trying to access the COM port. PlatformIO monitor, Arduino UI, Cura etc
1. Have you hit shift+F5 on the web interface in case some old data is being cached?
1. Is there a pulsing circle in the bottom right of the device?
1. Is the COM device appearing with the VID and PID you expect?
[image]

i-am-shodan, 2025-01-08:
I've also tested the latest agent dll from the latest build and it works fine

i-am-shodan, 2025-01-08:
The agent hasn't been installed correctly and isn't running. Did you use the us-autorun.ds script?

gliderexpert, 2025-01-08:
Yes, that's the one I used.

i-am-shodan, 2025-01-08:
Can you execute this script for me
```
GUI R
DELAY 1000
STRINGLN notepad
STRINGLN VID = #_VID_
STRINGLN PID = #_PID_
ENTER
STRING cmd /c @echo off & for %d in (D E F G H) do if exist %d:\in1.bat %d:\in1.bat #_VID_ #_PID_
```

and post the output

gliderexpert, 2025-01-08:
Sure. I saved it as script.ds, placed it on the SD card, launched it through the web control panel and... nothing happens :)
By the way, the logs show that it executed:
Run file/script.ds  
Setting payload to - '/script.ds'  
Script finished execution

i-am-shodan, 2025-01-08:
What happens when you do this:
https://github.com/i-am-shodan/USBArmyKnife/issues/35#issuecomment-2577892206

gliderexpert, 2025-01-08:
The Start menu in Windows opens.

i-am-shodan, 2025-01-08:
Can you issue each one of the commands in the script

i-am-shodan, 2025-01-08:
Are you creating these scripts on Linux?

gliderexpert, 2025-01-08:
on windows

i-am-shodan, 2025-01-08:
Can you attach the exact script you created

gliderexpert, 2025-01-08:
of course
https://mega.nz/file/uQYSxI5b#HajUzMt8St4PKzxxvaFluUK7qEPy0EDVHuXMf-KdZaE

gliderexpert, 2025-01-08:
If I run these commands one by one, everything goes smoothly (instead of #_VID_ and #_PID_, the actual PID & VID values of the device are substituted), up until pressing Enter. It simply doesn’t work because there’s no space, so it launches notepadVID=....

i-am-shodan, 2025-01-08:
I've found some bugs to fix. I will have a think about your problem and push some fixes to what i've found that maybe will help.

gliderexpert, 2025-01-08:
Okay, thanks! I'll be looking forward to it.

inv1sible, 2025-01-08:
THANKS for this issue. It helped me understanding better how an agent works and getting an agent running.

i-am-shodan, 2025-01-09:
I have created a wiki page here, https://github.com/i-am-shodan/USBArmyKnife/wiki/Agent. Feel free to expand it

i-am-shodan, 2025-01-09:
@gliderexpert I think you've got your machine and the device in a bit of a state with all the testing. I have written some steps for you to try. Perform each step in turn and STOP if you see an error or the state of your machine does not reflect what's written. It is important that you follow these steps accurately.

1. Remove every last trace of the agent on your device following the instructions here https://github.com/i-am-shodan/USBArmyKnife/wiki/Agent#how-can-it-be-removed
1. Erase your SD card and create a new <32gb partition. Do NOT put any files on the SD card.
1. Erase your device and reflash using web browser method, instructions are [here](https://github.com/i-am-shodan/USBArmyKnife/wiki/Installation#flashing-with-a-web-browser)
1. Insert your SD into the device and plug in
    * Confirm you have a pulsing circle in the bottom right of the device's screen
1. Connect to the web interface and set your VID & PID to the values you've used previously
1. Unplug the device and plug in again
    * Confirm a new HID device has appeared
    * Confirm a new Serial device has appeared and that the serial device has the VID and PID you specified in the previous step
    * Confirm you can connect to the serial device with Putty and DONT get an access denied message. Close Putty.
1. Build a agent.img using the agent binaries from the last successful build I shared
1. Unplug the device and add the agent.img.
    * Your SD should ONLY contain two files autorun.ds & agent.img
1. Insert the SD card into your device and plug into your victim machine
    * Confirm you see keypresses that open the run dialog and type a command in
    * Do not remove the device
1. After 5 minutes confirm that:
    * 3 DLLs are in the %APPDATA% folder
    * The scheduled task that you deleted has returned
    * The VID and PID you set is visible as an argument to the scheduled task
    * The scheduled task is not marked as disabled
    * You have at least one dllhost.exe process running
    * You get an access denied message if you try and connect to the COM port with Putty
1. Unplug the device, insert the SD card into your machine and confirm you have a file named agentInstalled
1. Reinsert the SD card into the device and plug in
1. Navigate to the devices web interface using an in private tab to ensure nothing has been cached
1. Confirm the agent is marked as connected

gliderexpert, 2025-01-09:
Everything went fine up until step 15. When I connect the device, nothing happens. Also, it's worth noting that the agent binaries are present in the %APPDATA% folder.

i-am-shodan, 2025-01-09:
There is no step 15?

gliderexpert, 2025-01-09:
oh, im sorry, 9 step

i-am-shodan, 2025-01-09:
I'm sorry it seems there is something wrong with the machine you are trying to install onto.

If you can't see the HID attack but the files have appeared there are only two possible options:
* You didn't clean up the attack from a previous run as I detailed
* Your machine isn't working correctly. The run dialog and the cmd window are for some inexplicable reason not appearing and you can't effectively debug on this machine.

Maybe the issues you're experiencing will be fixed by some commit in the future but without more verbose information on your side there is nothing else I can help you with.

i-am-shodan, 2025-01-17:
Closing due to lack of updates