i-am-shodan/USBArmyKnife #36: bug with AtomS3U binaries !?

Darkcast, 2025-01-08:
Im running a mac
Flashed  the following version of the atom binaries using the web-ui``` https://github.com/i-am-shodan/USBArmyKnife/actions/runs/12656404064```

when I conenct to is I see this on the dashboard
[image]
and after a few seconds or as soon as I refresh it disconnects 
I cant get device logs because the device disconects or reboots
feels like the device wifi might be turn on and off ? 

let me know if there is something I need to try or redo

Thank you in advance :)

i-am-shodan, 2025-01-08:
The device is crashing trying to render the file 4.3.2.1:8080/data.json. This contains all the configuration information for the web UI.

I have not attempted to flash this specific device using the web ui. Last time I saw an issue like this it was a bad SD card.

i-am-shodan, 2025-01-08:
Are you able to try flashing it using the VS code method?

i-am-shodan, 2025-01-08:
Ok thinking about it I understand the problem. The S3U has a flash file system aka no SD card. The device is almost certainly trying to tell you via the serial port that the flash file system is corrupt as it hasn't been initiated. You can only do that via the VS code method.

I'll update the wiki.

i-am-shodan, 2025-01-08:
Updated wiki.

closed 2025-01-08 by i-am-shodan

Darkcast, 2025-01-08:
> Are you able to try flashing it using the VS code method? Otherwise I will try flashing it tomorrow and see if I can reproduce.

I havent tried that.

Darkcast, 2025-01-08:
thanks for the entry on the installation page but there no instruction on how to flash on AtomS3U :(

i-am-shodan, 2025-01-08:
If you try it you'll see there is an option for your device

Darkcast, 2025-01-09:
will do, thank you very much :)