i-am-shodan/USBArmyKnife #37: Evilap did not turn up

michael-fixics, 2025-01-09:
Hi,

i have tried to put the autorun.ds in the SD card.

And run the army knife as per video.

But no wifi with the name turned up on wifi scanning

Any idea ?

i-am-shodan, 2025-01-09:
What device are use using?
Did any WiFi AP appear?
What if you don't use the script?

michael-fixics, 2025-01-09:
1. LilyGo-S3 dongle
2. None. the usual SSID 'iPhone14' also didn't show up on wifi scan
3. When I remove autorun.ds, then replug the dongle, iPhone14 would then show up

i-am-shodan, 2025-01-09:
Are you upgrading from a previous install?

michael-fixics, 2025-01-09:
i did a full clean. then upload in VS code.

i-am-shodan, 2025-01-09:
Can you delete the USB Army Knife folder and start again from scratch
OR
Can you use the new web flashing route with the latest firmware binary images (see wiki)

I think that will solve the issue

i-am-shodan, 2025-01-09:
I'll check the build actions and confirm it's pulling the latest esp32 marauder code. If you're saying it doesn't work with the binaries you're downloading this might be the issue

i-am-shodan, 2025-01-09:
I've updated the github action, @jimjimminy @michael-fixics would you mind testing this and see if it works

michael-fixics, 2025-01-10:
it worked after i used web flashing.

Great work here ! ; )