i-am-shodan/USBArmyKnife #40: Can't write file to sdcard on ESP32-S3

kevin-fibich, 2025-01-17:
Trying out the coverstorage example on Waveshare ESP32-S3 1.47inch, i got an error in the Log "Script error near line 14", which is the CREATE_FILE() command and the file is never created. Uploading a testfile via web interface gives no error and a file with correct name but empty content is created.

Seems to me something in ESP vfs handling is broken.

I tried formatting the sdcard in Windows and with the included format function, both result in the same problem.
Switching to a different sdcard (first was noname, second Sandisk) also did not help.

Reading files and executing scripts put on the sdcard via windows works fine.

i-am-shodan, 2025-01-17:
I think i've found the bug
https://github.com/i-am-shodan/USBArmyKnife/blob/77530f6c974c8c44212a8c75b99f9bfff8c71fd4/src/Devices/Storage/HardwareStorage.cpp#L230

This line should use "w" instead of "r". If you are compiling yourself can you try this change and see if the script works?

kevin-fibich, 2025-01-17:
I used precompiled firmware but i will set up to compile myself later this afternoon and try it.

kevin-fibich, 2025-01-17:
Would this also explain the empty content on uploads via web ui?

kevin-fibich, 2025-01-17:
okay, firstrun file is created now, but i dont see a storage device on either the first and the second time i connect.

kevin-fibich, 2025-01-17:
okay, with a different sdcard it works now! Your patch is correct.

i-am-shodan, 2025-01-17:
Are you still having issues with the web UI uploading files?

kevin-fibich, 2025-01-17:
Right now i cant use the web UI as it seems to reset as soon as i connect to wifi.

i-am-shodan, 2025-01-17:
I don't use that device very much, i'll try and check with it to see if i've introduced a bug

i-am-shodan, 2025-01-21:
I've just run through some tests with this device, I can't see any issues my side. If you still have issues try the latest release and reoped.

closed 2025-01-21 by i-am-shodan