i-am-shodan/USBArmyKnife #44: FEATURE: Wardriving Mode + BLE SPAM

corrupt3k, 2025-01-24:
It would be cool to see something similar to the pwnagotchi project where you can collect handshake PCAPs and send them to wpa-sec.stanev.org for cracking. Maybe even have the screen update on what it has found.

In tandem, an option to do BLE SPAM as well to freak out nearby mobile devices.

i-am-shodan, 2025-01-24:
Most of that functionality is already there. You can BLE spam and collect WiFi Auth packets with the marauder extension command. You do need to grab the pcap files from the web server yourself tho.

CDRLpart, 2025-01-26:
I think the person wants you to do a Bluetooth attack with Bluetooth advertising so that e.g. not your airpods pops up or something like that on iPhones

i-am-shodan, 2025-01-26:
Check the wiki, USB Army Knife supports most ESP32 Marauder attacks including those for WiFi and Bluetooth.

It sounds like these commands will do what you want.

```
ESP32M btwardrive
ESP32M blespam -t all
```

Read the docs here

https://github.com/justcallmekoko/ESP32Marauder/wiki/bluetooth-wardrive

https://github.com/justcallmekoko/ESP32Marauder/wiki/bt-spam-all

i-am-shodan, 2025-01-26:
Other examples for WiFi auth collection https://github.com/i-am-shodan/USBArmyKnife/issues/48

closed 2025-01-26 by i-am-shodan

i-am-shodan, 2025-01-29:
https://github.com/i-am-shodan/USBArmyKnife/blob/master/examples/wifi_deauth_and_crypt_capture/README.md