i-am-shodan/USBArmyKnife #45: usb_ethernet_pcap showing empty pcap file

pytagorics, 2025-01-24:
Hi! I got my LilyGo T-Dongle S3 and I´m playing around with the project. 

The issue here is that when running the usb_ethernet_pcap I get an empty pcap file, I´ve tried some times in different devices with the same result.

I just run the default autorun file available in the Github.

Could you please guide me on how to make this work? Or if you´ve seen anything like it before?

I can provide more details if necessary.

Thanks in advance

Cheers

P.D. Your project is awesome btw!

i-am-shodan, 2025-01-24:
* Have you set the mode to be USB NCM like the docs and the video say. This involves changing a setting in the web interface.
* Does a new USB Ethernet adapter appear in the network connections window? Send me a picture
* Send me the logs from the device

i-am-shodan, 2025-01-24:
Delete the autorun.ds file, unplug the device and plug in again then send me a picture of the screen.

pytagorics, 2025-01-24:
Thanks for the reply!

I did set the mode to USB NCM as the video.

I´ll collect the info and I´ll posted here

pytagorics, 2025-01-24:
[image]
[image]
[image]
[image]
[image]
[image]

pytagorics, 2025-01-24:
I paste all the screenshots requested... after deleting the autorun file I connected the lilygo and the screen said 

Device now running
USB MODE: NCM
USB CLASS: None

i-am-shodan, 2025-01-24:
Can you show me a screenshot of your Windows network connections window.

pytagorics, 2025-01-24:
[image]

pytagorics, 2025-01-24:
This one is with the dongle connected.. Actually before connecting it was the same

pytagorics, 2025-01-24:
In the network connections window one of the devices should say tinyusb in it. If you see that device In what IP address does it have?

- I don´t see it

pytagorics, 2025-01-24:
[image]

i-am-shodan, 2025-01-24:
OK so the problem here is that the device is expressing itself as a USB NCM device but your guest OS isn't receiving that. I don't know the inner workings of Virtual Boxe's USB stack but I suspect the problem is there.

Maybe virtual box doesn't support USB NCM.

pytagorics, 2025-01-24:
Just to add I´m working from my host not from the Virtual Box

So do you think is virtual box is causing the issue?

I will uninstall vbox and retry

Or do you think trying in a device that doesnt have any virtualization software would work?

i-am-shodan, 2025-01-24:
OK,

Do you have anything under the USB devices entry in device manager that has a problem?

pytagorics, 2025-01-24:
[image]

pytagorics, 2025-01-24:
I only have this

i-am-shodan, 2025-01-24:
Have any other USB Army Knife attacks worked? Or have you just tried this one?

pytagorics, 2025-01-24:
evilap worked perfectly

pytagorics, 2025-01-24:
just tried again and worked fine

pytagorics, 2025-01-24:
Do I need to introduce any network credentials or so? Or the script won´t need any? How are the internals of the script working?

I have absolutely no clue hence the question

i-am-shodan, 2025-01-24:
I've just flashed the latest release and checked USB NCM support works. It does for me.

pytagorics, 2025-01-24:
Mmmm... Do you have any clue of why is not working for me?

Do you have any virtualization software installed?

What can you see in the network connections?

i-am-shodan, 2025-01-25:
When I plug the device in:
* I see two new devices in device manager. One is under networking and another under USB
* In the network connections window another device appears.

If you aren't seeking my those things it's not related to the device but the operating system.

Has VirtualBox taken this USB device and passed it through to a VM? I know VMware used to work like that.

pytagorics, 2025-01-25:
Hi! Sorry for late replay.. Today has been a busy day

I connected the lilygo to a device that doesnt have any virtualización software installed or anything, Windows 8.

When connected the lilygo into the win8 device, a new device appeared in device manager called tinyusb but no other device appeared in device manager and in network connections I just had the iPhone14 network.

When I checked the pcap from this test It was empty again.

Does the traffic have to pass through certain specific network or just wifi is ok?

What do you think of this?

Btw, thanks a lot for your help!!!

i-am-shodan, 2025-01-26:
Great it sounds like the device is working. I'm not sure if windows 8 supports USB NCM network drivers bc it's too old.

Pcapping the USB Ethernet packets requires that network traffic goes through usb Ethernet. To pcap the WiFi you need to use the usb marauder extension commands.

Sounds like you should investigate virtual box + usb.

pytagorics, 2025-01-26:
_Great it sounds like the device is working. I'm not sure if windows 8 supports USB NCM network drivers bc it's too old._

-it's great is working indeed! I think I could get a win10 device without any virtualization software in it, I'll do the testing again on that one and I'll keep you posted.

_Pcapping the USB Ethernet packets requires that network traffic goes through usb Ethernet._

-How can I configure this when connecting the device? Is this the traffic generated when I connect to the iphone14 network?

-could you please guide me on this?

_ To pcap the WiFi you need to use the usb marauder extension commands._

-Awesome! Is this something the project already has developed or support?

_Sounds like you should investigate virtual box + usb._

-Yeah, definitely.. I'll get deep in it and whatever I find I'll let you know in case is useful for the project

i-am-shodan, 2025-01-26:
The WiFi network and the USB Ethernet network are completely separate.

If you plug in the device and it's correctly detected as USB NCM then the host will automatically send network traffic. In the few seconds my example is pcaping I get a 59kb file.

Sniffing the air interface works by using the ESP32 Maurader command set. Check the wiki for more details. There is a command https://github.com/justcallmekoko/ESP32Marauder/wiki sniffraw that should do that

closed 2025-01-26 by i-am-shodan