i-am-shodan/USBArmyKnife #46: Reset Wifi Password

marcoliverhofer, 2025-01-25:
Hi,

there might be an easier way to reset the Wifi Password, I just can't figure it out.
Any idea ? Or do I have to re-flash the original firmware, re-install USBArmyKnife ?

Thanks. Great work by the way, really well done Project.

i-am-shodan, 2025-01-25:
* Go to the web interface
* click on setting
* find the WiFi password setting and set it

marcoliverhofer, 2025-01-25:
Right. 

Problem is, I can't connect anymore to the stick's Wifi as I am an idiot and somehow forgot that I changed the password from the "original". So, I can't even reach the web-interface of the "web server" on the stick :/ anymore to make any changes.

i-am-shodan, 2025-01-25:
You options include:
* completely erasing the flash, then reflashing with the code. Only the VSCode method has the option for this
* changing the password again but this time with a script using the brand new settings modification commands. You'd need to get the ID of the password setting which you can get from the source

marcoliverhofer, 2025-01-25:
Ah thanks, as I thought. 

I already used the VSC to get this thing going, so will go down this path first. But will look into the second option as well, haven't thought about it. Might come in quite handy later on.

i-am-shodan, 2025-01-25:
Untested, might work

SET_SETTING_STRING wifi-pwd password

closed 2025-01-25 by marcoliverhofer