i-am-shodan/USBArmyKnife #48: Feature requests: DeAuth Attack for collecting WiFi hashes

CDRLpart, 2025-01-26:
I think it would be very cool to be able to kick everyone out of the other Wifi if there's a second wifi with the same name to collect the WiFi Password hashes or passwords from a wifi. I mean on Kali Linux this IS possible but with this tiny device IT would be much more powerful
Thanks 😃👍

i-am-shodan, 2025-01-26:
You can already do this. Look at the esp Marauder deauth commands/auth pcap command in the wiki

closed 2025-01-26 by i-am-shodan

i-am-shodan, 2025-01-26:
https://github.com/justcallmekoko/ESP32Marauder/wiki/deauthentication-attack-workflow

ESP32M attack -t deauth

i-am-shodan, 2025-01-26:
https://github.com/justcallmekoko/ESP32Marauder/wiki/sniffpmkid

i-am-shodan, 2025-01-29:
https://github.com/i-am-shodan/USBArmyKnife/blob/master/examples/wifi_deauth_and_crypt_capture/README.md

d4rkw1n9, 2025-04-16:
@i-am-shodan Not really worth of reopening the feature request again, but would love to know whether it is possible to save the captured handshakes with the respective AP in the filename? I mean, the tool goes rogue and attacks everything within it's reach as far as I understand. Therefore it would be great to have the captured files ordered according to the AP name - to be able to cherry pick the interesting ones. Thanks :)

i-am-shodan, 2025-04-16:
You just need to use ESP marauder commands to specify the access point you want to attack.

d4rkw1n9, 2025-04-16:
> You just need to use ESP marauder commands to specify the access point you want to attack.

Yes, but with the original autorun.ds from this repo, it will attack all I guess (which has its legit use case, ofc). In that case, it would be beneficial when the handshakes are saved with the AP in the file name.