i-am-shodan/USBArmyKnife #49: WINUSB Driver gets loaded on fresh windows install

corrupt3k, 2025-01-26:
Windows 11 24H2

Device USB\VID_303A&PID_1001&MI_02\6&366d4ba9&0&0002 was configured.
Driver Name: winusb.inf
Driver Package ID: winusb.inf_amd64_e3d2ed0c43a36b9a
Class GUID: {88bae032-5a81-49f0-bc3d-a4ff138216d6}
Driver Date: 06/21/2006
Driver Version: 10.0.26100.1150
Driver Provider: Microsoft
Driver Section: WINUSB.NT
Driver Rank: 0xFF2000
Matching Device ID: USB\MS_COMP_WINUSB
Outranked Drivers: 
Device Updated: false
Parent Device: USB\VID_303A&PID_1001\64:E8:33:6C:08:20

Device USB\VID_303A&PID_1001\64:E8:33:6C:08:20 was configured.
Driver Name: usb.inf
Driver Package ID: usb.inf_amd64_e6361bd9b1fba04a
Class GUID: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 06/21/2006
Driver Version: 10.0.26100.1882
Driver Provider: Microsoft
Driver Section: Composite.Dev.NT
Driver Rank: 0xFF2006
Matching Device ID: USB\COMPOSITE
Outranked Drivers: 
Device Updated: false
Parent Device: USB\ROOT_HUB30\4&2b4c5684&0&0

I have tried this on three machines now.

corrupt3k, 2025-01-27:
Plugged my T-Dongle in 10 times on the same machine.
4 of those times got the correct VID PID (VID_CAFE&PID_403E)

corrupt3k, 2025-01-27:
And now it works every time.
So strange.
Closing.

corrupt3k, 2025-01-27:
I'm still having problems here.
About 1 out 4 times I get the wrong VID PID

i-am-shodan, 2025-01-27:
What other vid/pid do you get?

If you enter DFU mode with the button held down you will get another pid. Is this what you did?

corrupt3k, 2025-01-27:
The VID PID that I get is the same one as when I put it in DFU mode. 303A 1001

It's feels like a weird failsafe to enter DFU mode without holding the button down.

i-am-shodan, 2025-01-27:
Well that VID/PID doesn't come from my code. I suspect you have something on your machine sending a resetDFU command to the device.

Not sure what this could be but things like Arduino, esptool, platformio and even cura have been known to do this.

i-am-shodan, 2025-01-28:
This really doesn't sound like a problem with this code base

closed 2025-01-28 by i-am-shodan