i-am-shodan/USBArmyKnife #53: No web interface shows up.

RapiidB, 2025-01-29:
Hello,

I have just installed the USB army knife on a lilygo T dongle S3. After inserting an empty formatted 32GB SD card, the USB shows

i-am-shodan, 2025-01-29:
Have you tried http://4.3.2.1:8080

RapiidB, 2025-01-29:
Thank you for your insanely quick response. Yes, ive tried that, however no result. Also tried flashing it with the other method (web, i tried the vscode install first) but it made no difference

i-am-shodan, 2025-01-29:
What devices have you tried to access the site on? If its a phone try turning cellular data & Bluetooth off

RapiidB, 2025-01-29:
Just checked your recommendation of turning off bluetooth and cellular but still not working. Also when i scan the iPhone14 network with a scanner like Fing, it shows the 4.3.2.1 device, but no open ports.

i-am-shodan, 2025-01-29:
I've assumed you've tried turning it off an on again. This is an embedded device you can hit it with a port scanner and expect it to survive the encounter. Have you tried flashing the build from the latest release with the web method. That is known to have no issues with the web server.

You should also try to access it with another device just in case it's your phone that's the problem.

closed 2025-01-29 by RapiidB