i-am-shodan/USBArmyKnife #54: FEATURE REQUEST: Marauder GUI and Script from web ui

CDRLpart, 2025-01-29:
FEATURE REQUEST, No Bug but I didnt got IT away: it would be very cool if you could also send the scripts via the web ui if you don't have it on the SD card and have a direct terminal for the esp32 marauder functions or even a GUI like when you connect one to the Flipper Zero

i-am-shodan, 2025-01-29:
I've considered this but its a lot of work. You can get around it today by authoring it on your device and uploading it via the web UI.

CDRLpart, 2025-01-29:
Okay thank u

CDRLpart, 2025-01-29:
Can you also simulate a webcam with a specific video via the USB port in USB Army Knife?

i-am-shodan, 2025-01-29:
Not today but it wouldn't be hard to add. There is already code here in the upstream project https://github.com/i-am-shodan/Adafruit_TinyUSB_Arduino/blob/master/examples/Video/video_capture/video_capture.ino

CDRLpart, 2025-01-29:
So you could implement that?

i-am-shodan, 2025-01-29:
Yes