i-am-shodan/USBArmyKnife #55: FEATURE: Pause a script

corrupt3k, 2025-01-29:
VAR $PAUSED = 0

WHILE ( TRUE )
	DISPLAY_CLEAR
	IF ($PAUSED == 1)
		DISPLAY_TEXT 0 35 Paused.
		WAIT_FOR_BUTTON_PRESS
		IF (BUTTON_SHORT_PRESS())
			$PAUSED = 0
		END_IF
	ELSE_IF ($PAUSED == 0)
		DISPLAY_TEXT 0 35 Waiting...
		**DELAY 60000 - WAIT FOR BUTTON PRESS FOR 60 SEC?
                                 IF (BUTTON_SHORT_PRESS())
			                 $PAUSED = 1
		                 END_IF**
		F5
		DISPLAY_CLEAR
		DISPLAY_TEXT 0 35 Key Pressed.
		DELAY 5000
	END_IF
END_WHILE

i-am-shodan, 2025-01-29:
That script is not valid, you must call `WAIT_FOR_BUTTON_PRESS` prior to `BUTTON_SHORT_PRESS()`. There will be a `WAIT_FOR_BUTTON_PRESS_OR_TIMEOUT 1000` in the next release

i-am-shodan, 2025-01-29:
Closing as didn't use template

closed 2025-01-29 by i-am-shodan