i-am-shodan/USBArmyKnife #57: Agent for Linux.

freitas666, 2025-01-31:
Device: lilygo dongle s3

Doubt its possible use agent on linux?

Actually works very well on windows... 

Tks

tbautista, 2025-02-01:
Hello freitas666, i am having problem making the agent work on windows. I can see it on task scheduler, I even modified the script to have my hardcoded VID and PID but the only thing I see on logs is "Requesting Wait for 2000"

i-am-shodan, 2025-02-01:
Closing this as your both not using GitHub issues correctly.

It's probably 1h works to get it working in Linux

Other guy try running this script instead of the agent. https://github.com/i-am-shodan/USBArmyKnife/blob/master/tools/DebugLogs/get_device_debug_output.ps1

closed 2025-02-01 by i-am-shodan