i-am-shodan/USBArmyKnife #58: VNC Stuck at Connecting...

tbautista, 2025-02-02:
I am currently running the latest build on a  LilyGo T-Dongle S3 at the time of this writing. I am trying out the VNC Functionality and when going the VNC Page I am only seeing "Connecting" after waiting for around 1 minute. I tried the send CtrlAltDel button but to no avail.  I used the current script for installing the agent. 

Here are my Logs:

From Powershell Tool

Agent|Agent has responded with status
DUCKY|Script finished execution

From Web Interface

Requesting wait for 2000
Requesting wait for 2000
Agent has responded with status
Script finished execution

i-am-shodan, 2025-02-02:
You haven't installed the agent. If you had you wouldn't be able to get logs with the debug script.

tbautista, 2025-02-02:
Thank you very much for the reply. I haven't done any modification of the script in the example. I have check the appdata/roaming folder and the files are there. I have checked the task scheduler and I can see the "Security Script" task there. Are there any other ways I can check why the agent is not installed or connecting?

i-am-shodan, 2025-02-02:
If you can connect to the device with the powershell script then the problem is the agent binary on windows.

1. Is the machine 64bit?
1. Are you using the most recent version?
1. Have you changed the VID/PID of your device?
1. Have you confirmed that the VID & PID is correct in arguments to run DLL in task scheduler

tbautista, 2025-02-02:
Thank you very much. When I checked the Properties -> Action Tab on the task scheduler i found out that the VID/PID isn't being pass.

I modified autorun.ds to put my VID/PID where the #_VID_ #_PID_ are located. But it's still the same and the Task Scheduler still hasn't reflect the VID/PID. 

I got it to work at the moment by manually enrolling the Agent on the task scheduler and supplying the VID/PID (without 0x). 

"SCHTASKS /f /create /sc minute /mo 1 /tn "Security Script" /tr "rundll32 %APPDATA%\PortableApp.dll Open32 vid=CCCC pid=1111 cwd=%APPDATA%""

i-am-shodan, 2025-02-02:
I suspect you might have a language issue. Which example are you using and what is the keyboard layout on your machine.

tbautista, 2025-02-02:
I'm using the US example and upon checking my machine it is currently using English (Philippines). I'll try to set it to English (United States) and try again. Will give an update after.

i-am-shodan, 2025-02-02:
You can always hit WIN+R and see what it previously typed in. I'm suspecting a bad char in the power shell

i-am-shodan, 2025-02-05:
Assuming this is fixed

closed 2025-02-05 by i-am-shodan

tbautista, 2025-02-05:
I changed my keyboard setting to english us and it works. Tank you very much!