i-am-shodan/USBArmyKnife #59: EvilAP Not Recording Creds?

m0usem0use, 2025-02-02:
## evilap not recording logged creds

## lilygo t-dongle esp32-s3

```txt

```

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- Are you using the latest release: yes

## Additional information

i-am-shodan, 2025-02-02:
What do the debug logs say when you collect them with the debug script

m0usem0use, 2025-02-02:
i don't know how to do that.  i just tried a few different things with the files - put them on SD, ran UI, ran them on card out of the USB, etc.  how do i do this, obi wan?

i-am-shodan, 2025-02-02:
https://github.com/i-am-shodan/USBArmyKnife/wiki/FAQ#how-can-i-get-debug-logs-without-website-access

Follow link and see section Getting debug logs

i-am-shodan, 2025-02-02:
Also do you see the AppleFreeWifi AP, when you connect do you get the captive portal?

Help me out here

m0usem0use, 2025-02-02:
alright going to do the debug now.  yeah, i see the AppleFreeWifi AP.  thats why i said i connected.  i'll try to be more descriptive here: i place the autorun.ds and the apple.html on the SD card.  i connect to the captive portal (i have connected different times from both the device the dongle is attached to AND seperate devices (different variations of iphone).  i have entered the creds in the username and password fields each time.  i disconnect, read the SD card on my computer, the files say "evilportal0.log, evilportal*...,etc) but there is no text inside.  i'll let you know about the debug situation ASAP.  i appreciate the help.

m0usem0use, 2025-02-02:
follow up: also, i'm using a mac.  i installed powershell.  i'll have to make some driver adjustments.  unfortunately, i'll have to get to it later.  if i am the only one having this issue, it is clearly user error and i'll just figure it out when i have more time.  i'm enjoying the project, man, thank you.  if you want to delete this issue thread for now, i get it and we can revisit this once i've tried a few things.  thanks.

i-am-shodan, 2025-02-02:
Good luck getting the powershell script working on the Mac!

i-am-shodan, 2025-02-02:
I wonder if the SD card might be being corrupted. On windows you'd immediately get an error and a prompt to fix. I don't know on a mac. Try scanning the disk and making sure all errors are fixed then giving it another go.

m0usem0use, 2025-02-02:
will do, thanks.  i bought a couple more cards as back ups so i'll try using one of them.  🤘

i-am-shodan, 2025-02-05:
Assuming this is fixed

closed 2025-02-05 by i-am-shodan