i-am-shodan/USBArmyKnife #62: Ducky layout not work

SnowGlobeio, 2025-02-04:
## Your problem

I tried to set the "win_fr" in a script.
For that, i enabled "-D LOCALE_win_fr" in platformeio.ini and update the LiliGo T-dongle with this new firmware and set "KEYBOARD_LAYOUT win_fr" in my script.
The script stop in log just after setting the layout.
If i test without the line "KEYBOARD_LAYOUT win_fr" the script work fine (but in english...) 

## Your device

LilyGo T-Dongle S3

# Language (HID attacks only)

## Your DuckyScript

```
REM Test V1.1
LED_OFF
DISPLAY_CLEAR
KEYBOARD_LAYOUT win_fr

GUI R
DELAY 800
STRING powershell -NoP -NonI -Ep Bypass
DELAY 800
ENTER
DELAY 1200

DISPLAY_TEXT 0 20 Test Finish !!

```

## Your debug logs

```
Run file/test.ds
Setting payload to -'/test.ds'
Keyboard layout set to win_fr

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-04:
Can you uncomment the following line and flash this to the device.
```
;-D DUCKY_CUSTOM_LOG
```

This will add even more debugging from the DuckyScriptInterpreter project.

Systemac, 2025-02-05:
Same here...

I'm making a python script to modify the ducky scripts on the fly according to the layout, but it's not very practical...
That's why I sent you a message yesterday asking for a way to implement on-the-fly layout changes via the web interface...

I'm making a python script to modify the ducky scripts on the fly according to the layout, but it's not very practical...
That's why I sent you a message yesterday asking for a way to implement on-the-fly layout changes via the web interface...

i-am-shodan, 2025-02-05:
I'll test today and see if I can reproduce

i-am-shodan, 2025-02-05:
First I download the latest code, then I did a 'Full Clean' to ensure nothing of any old build could remain. Then I changed the platformio.ini file to look like this. Once complete I clicked upload to make sure the latest code was flashed.

```
[keyboardlayouts]
build_flags = 
	; Select the keyboard layouts (if any) you want available by uncommenting the line
	; -D LOCALE_win_be
	; -D LOCALE_win_ca_FR
	; -D LOCALE_win_ca
	; -D LOCALE_win_ch
	; -D LOCALE_win_cs_CZ
	; -D LOCALE_win_de_DE
	; -D LOCALE_win_da_DK
	;-D LOCALE_win_en_GB
	; -D LOCALE_win_es
	; -D LOCALE_win_es_MX
	; -D LOCALE_win_fi
	-D LOCALE_win_fr
	; -D LOCALE_win_hr_HR
	; -D LOCALE_win_hu_HU
	; -D LOCALE_win_it
	; -D LOCALE_win_ja_JP
	; -D LOCALE_win_no
	; -D LOCALE_win_pt_BR
	; -D LOCALE_win_pt_PT
	; -D LOCALE_win_se
	; -D LOCALE_win_si
	; -D LOCALE_win_sk_SK
	; -D LOCALE_win_tr_TR
```

I had no autorun.ds on my SD, I then used this script which I called test.ds and launched from the web UI

```
REM Test V1.1
LED_OFF
DISPLAY_CLEAR
KEYBOARD_LAYOUT win_fr

GUI R
DELAY 800
STRING powershell -NoP -NonI -Ep Bypass
DELAY 800
ENTER
DELAY 1200

DISPLAY_TEXT 0 20 Test Finish !!
```

Just before I did this I ran `.\get_device_debug_output.ps1` to get debug logs as soon as the device starts executing the script. This is what I got:
```
Agent|Agent has responded with status
WEB|Run file/test.ds
DUCKY|Setting payload to - '/test.ds'
DUCKY|Keyboard layout set to win_fr
DUCKY|Requesting wait for 800
DUCKY|Requesting wait for 800
DUCKY|Requesting wait for 1200
DUCKY|Script finished execution
```

I saw the Windows run box appear and a PowerShell command (with incorrect keys - I'm still using my keyboard layout). I also saw the devices screen clear and then the message `Test Finish !!` appear on screen.

I've proved it works. Here is a ZIP with the firmware I used and the exact DuckyScript file. You can flash this fw to your device if you use some of the files from the last successful build and the instructions on the wiki.

[t.zip](https://github.com/user-attachments/files/18671050/t.zip)

SnowGlobeio, 2025-02-05:
Why have you disabled "win_en-GB" in platformio.ini ?!
Let's set in available, and you will see that it will not work.

If you select only "win_fr" you don't need to set "KEYBOARD_LAYOUT win_fr" in script because it's the only available in the firmware.

i-am-shodan, 2025-02-05:
That is incorrect. The only available is en-US which is the USB standard.

Systemac, 2025-02-05:
> First I download the latest code, then I did a 'Full Clean' to ensure nothing of any old build could remain. Then I changed the platformio.ini file to look like this. Once complete I clicked upload to make sure the latest code was flashed.

I haven't tried your solution yet, but how do you do a full-clean???

SnowGlobeio, 2025-02-05:
@Systemac write :  After testing, activating only this line (win_fr) does indeed work...

So, it's apear impossible to activate several layout and select them in the ducky script.

i-am-shodan, 2025-02-05:
I think this is a RAM issue.

The tables can get quite large and currently only live in main RAM. The device is already storing the full USA HID table, two more is currently too many.

They'll need to be moved to PROGMEM.

Systemac, 2025-02-05:
So I've just tested two tables, win_fr and win_en_GB, the script with KEYBOARD_LAYOUT win_fr works, but with KEYBOARD_LAYOUT win_en_GB doesn't, the logs tell me that the layout hasn't been found, In this case, it may not be a ram problem, as I didn't get any compile errors, and logically, win_en_GB loads before win_fr, or maybe KEYBOARD_LAYOUT win_en_GB isn't the right wording...

i-am-shodan, 2025-02-06:
As the wiki says its `KEYBOARD_LAYOUT win_en-GB`

en-GB is the correct format for a locale but you cant have all those chars in a #define

i-am-shodan, 2025-02-06:
I've just tested with two tables compiled in win_fr and win_en_GB. This allows you to target three languages:
* win_en-US - this is the default for USB HID and must always be included
* win_en-GB - UK keyboard layout
* win_fr - French keyboard layout

I was able to run the script and it all worked fine. As such I do not believe there is an issue with the code.

There is an upper limit to the number of tables you can include. The more you add the more RAM it requires and for an embedded device RAM is in short supply. Maybe you decided to uncomment all lines - that would be bad.

Systemac, 2025-02-06:
OK, so it works, but I'm going to dig into the code to see if it's possible to make the change I wanted, i.e. modify the layout directly from the web interface.

i-am-shodan, 2025-02-06:
Am I missing something? Go into the web interface and type `KEYBOARD_LAYOUT win_en-GB` into the RAW box. Job done. This will effect all subsequent scripts that you run.

Systemac, 2025-02-06:
It's true that I hadn't thought of that, but I saw it more as a multiple choice in the parameters, with only the layout loaded during flash as a choice.

i-am-shodan, 2025-02-06:
As we've got to the bottom of this and you have a work around i'm closing this issue.