i-am-shodan/USBArmyKnife #64: REQUEST: ESPeFuse Summary for LilyGo T-Dongle S3

corrupt3k, 2025-02-05:
## Your problem

Could you post the output of your LilyGo T-Dongle S3 eFuse Summary?
I have a feeling I need to burn some eFuses to get things working right.
One of them being USB_PHY_SEL

## Your device

```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- I confirm I have read the FAQ: Yes
- I confirm I am using the latest release: Yes

## Additional information

corrupt3k, 2025-02-06:
Here is my output for reference.

```espefuse.py v4.8.1
Connecting...
Detecting chip type... ESP32-S3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
K_RTC_LDO (BLOCK1)                                 BLOCK1 K_RTC_LDO                                   = 28 R/W (0b0000111)
K_DIG_LDO (BLOCK1)                                 BLOCK1 K_DIG_LDO                                   = 0 R/W (0b1000000)
V_RTC_DBIAS20 (BLOCK1)                             BLOCK1 voltage of rtc dbias20                      = 48 R/W (0x0c)
V_DIG_DBIAS20 (BLOCK1)                             BLOCK1 voltage of digital dbias20                  = -28 R/W (0x87)
DIG_DBIAS_HVT (BLOCK1)                             BLOCK1 digital dbias when hvt                      = -36 R/W (0b11001)
ADC2_CAL_VOL_ATTEN3 (BLOCK1)                       ADC2 calibration voltage at atten3                 = -12 R/W (0b100011)
TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -5.0 R/W (0b100110010)
OCODE (BLOCK2)                                     ADC OCode                                          = 71 R/W (0x47)
ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = -84 R/W (0x95)
ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = -12 R/W (0b100011)
ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = 96 R/W (0b011000)
ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = 124 R/W (0b011111)
ADC2_INIT_CODE_ATTEN0 (BLOCK2)                     ADC2 init code at atten0                           = -160 R/W (0xa8)
ADC2_INIT_CODE_ATTEN1 (BLOCK2)                     ADC2 init code at atten1                           = -20 R/W (0b100101)
ADC2_INIT_CODE_ATTEN2 (BLOCK2)                     ADC2 init code at atten2                           = 68 R/W (0b010001)
ADC2_INIT_CODE_ATTEN3 (BLOCK2)                     ADC2 init code at atten3                           = 92 R/W (0b010111)
ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = 480 R/W (0x78)
ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = 504 R/W (0x7e)
ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = 452 R/W (0x71)
ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = -20 R/W (0x85)
ADC2_CAL_VOL_ATTEN0 (BLOCK2)                       ADC2 calibration voltage at atten0                 = 444 R/W (0x6f)
ADC2_CAL_VOL_ATTEN1 (BLOCK2)                       ADC2 calibration voltage at atten1                 = 228 R/W (0b0111001)
ADC2_CAL_VOL_ATTEN2 (BLOCK2)                       ADC2 calibration voltage at atten2                 = 0 R/W (0b1000000)

Config fuses:
WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 0 R/W (0b0000000)
DIS_ICACHE (BLOCK0)                                Set this bit to disable Icache                     = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Set this bit to disable Dcache                     = False R/W (0b0)
DIS_TWAI (BLOCK0)                                  Set this bit to disable CAN function               = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disable app cpu                                    = False R/W (0b0)
DIS_DIRECT_BOOT (BLOCK0)                           Disable direct boot mode                           = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Set the default UART boot message output mode      = Enable R/W (0b00)
PIN_POWER_SELECTION (BLOCK0)                       Set default power supply for GPIO33-GPIO37; set wh = VDD3P3_CPU R/W (0b0)
                                                   en SPI flash is initialized
PSRAM_CAP (BLOCK1)                                 PSRAM capacity                                     = None R/W (0b00)
PSRAM_TEMP (BLOCK1)                                PSRAM temperature                                  = None R/W (0b00)
PSRAM_VENDOR (BLOCK1)                              PSRAM vendor                                       = None R/W (0b00)
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_SYS_DATA2 (BLOCK10)                          System data part 2 (reserved)
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash fuses:
FLASH_TPUW (BLOCK0)                                Configures flash waiting time after power-up; in u = 0 R/W (0x0)
                                                   nit of ms. If the value is less than 15; the waiti
                                                   ng time is the configurable value.  Otherwise; the
                                                    waiting time is twice the configurable value
[... cut ...]
                                                    eFuse to configure VDD_SPI

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
                                                   ock cycle

Flash voltage (VDD_SPI) determined by GPIO45 on reset (GPIO45=High: VDD_SPI pin is powered from internal 1.8V LDO
GPIO45=Low or NC: VDD_SPI pin is powered directly from VDD3P3_RTC_IO via resistor Rspi. Typically this voltage is 3.3 V).```

i-am-shodan, 2025-02-16:
I suspect this might be related to:

https://github.com/i-am-shodan/USBArmyKnife/issues/74

i-am-shodan, 2025-02-17:
```
=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = -4.3 R/W (0b100101011)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = 420 R/W (0x69)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = -204 R/W (0xb3)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -236 R/W (0xbb)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = -216 R/W (0xb6)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = 284 R/W (0x47)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = -200 R/W (0xb2)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -196 R/W (0xb1)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = -176 R/W (0xac)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = -88 R/W (0b110110)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = 40 R/W (0b001010)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = 40 R/W (0b001010)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = -60 R/W (0b101111)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = -40 R/W (0b101010)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = -12 R/W (0b100011)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = -28 R/W (0b100111)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 64 R/W (0b010000)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Disabled R/W (0b11)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = False R/W (0b0)
                                                    boot mode
DISABLE_WAFER_VERSION_MAJOR (BLOCK0)               Disables check of wafer version major              = False R/W (0b0)
DISABLE_BLK_VERSION_MAJOR (BLOCK0)                 Disables check of blk version major                = False R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 0 R/W (0x00000000)
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = False R/W (0b0)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) determined by GPIO45 on reset GPIO45=Low or NC: VDD_SPI pin is powered directly from
```

closed 2025-02-17 by i-am-shodan

corrupt3k, 2025-02-17:
Burning the USB_PHY_SEL eFuse got things working right for me. It is interesting that yours does not have that fuse burned.
I wonder if there are different hardware revisions?