i-am-shodan/USBArmyKnife #65: USB Device type to None break Lilygo T-dongle S3

SnowGlobeio, 2025-02-06:
## Your problem

When in your script you set "ATTACKMODE HID STORAGE" the log return "Device started with USB Device type NOT set to None, this is recommended when using ATTACKMODE command"
So, from the web SETTING page you can set usbDeviceType to custom value 'None' and click Save.

If you disconnect your LilyGo T-Dongle S3 and reconnect it, it stop booting properly and the screen flashes black and white repeatedly.
No web interface either when this happens.

You have to erase completly your dongle and upload a new firmware.
## Your device

```
 LilyGo T-Dongle S3

```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-06:
* If the SD card isn't in the device does the device still crash?
* If the SD card is in the device, but there is no autorun.ds file, does the device still crash?

SnowGlobeio, 2025-02-06:
If the SD card isn't in the device does the device still crash? : Yes

If the SD card is in the device, but there is no autorun.ds file, does the device still crash? Yes

https://github.com/user-attachments/assets/c47da448-d255-4318-b8ff-3601382ace8f

lukeswitz, 2025-02-06:
Stuck at this same issue....

mpaw, 2025-02-09:
Just ran into the same problem. Would probably be good to add support for resetting the settings without having to flash it. Maybe by adding a `RESET` file to the root of the SD card, and if it sees it on boot, it resets settings back to default.

i-am-shodan, 2025-02-10:
This should be fixed in the most recent release

closed 2025-02-10 by i-am-shodan