i-am-shodan/USBArmyKnife #66: Request: Precompiled .net image or arm64 support

lukeswitz, 2025-02-07:
## Your problem

- Can’t compile the agent using arm64. M1+ Macs aren’t able to run a VM of windows that supports the x64 build instructions. 

Adding arm64 support to the agent or a prebuilt image would be great. 

## Your device

```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- I confirm I have read the FAQ: Y
- I confirm I am using the latest release: Y

## Additional information

i-am-shodan, 2025-02-07:
There is already a prebuild copy of the agent available - check the README.md in the tools dir

closed 2025-02-07 by i-am-shodan

lukeswitz, 2025-02-07:
Totally missed that. Thank you! Love the work