i-am-shodan/USBArmyKnife #67: Example scripts missing and vscode flash not working

Besix2, 2025-02-09:
## Your problem

I tried to install by using platformio on vscode. Everytime I try to upload I get this error:
`MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package`
I tried reinstalling. Clearing cache, deleting _/packages_
Website install works but /autorun.ds cant be opend and all the example scripts are missing. I tried placing the example folder on the sd card but that didnt work either. It only created mutliple autorun.ds scripts.

## Your device

  ```
LilyGo T-Dongle S3
```

# Language (HID attacks only)

//
## Your DuckyScript

//
## Your debug logs

```
`Running!
Could not open file: /autorun.ds`

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-09:
Sounds like you got it working but your SD card doesn't have a file called autorun.ds in the root. If your USB drive is D:\ the code is looking for D:\autorun.ds

You can't just copy the examples folder over and expect something to happen. Copy the contents of **just one** of the examples over and make sure you read the instructions for each example because there are prerequisites.

Besix2, 2025-02-09:
Website method works with files-I was just to stupid.

closed 2025-02-09 by Besix2