i-am-shodan/USBArmyKnife #70: Wrong hid code for & and STR HOME in german Layout, File upload system not really working.

Besix2, 2025-02-09:
## Your problem

The hid symbol for & is wrong in german Layout. It displays a / instead.
The Combination of STR HOME doesnt work. Probably because hid code for HOME is wrong.
Can it be that the delay times are not as long as with original rubber ducky?
File system upload clears my .ds file when uploading and I also display file option isnt working at all.

## Your device

```LilyGo T-Dongle S3

```

# Language (HID attacks only)

Configured to use German.
Victim has german Layout.

## Your DuckyScript

[script](https://github.com/hak5/usbrubberducky-payloads/blob/master/payloads/library/prank/The_Matrix-Wake_Up/payload.txt)

## Your debug logs

```
there arent any error logs.

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-09:
> The hid symbol for & is wrong in german Layout. It displays a / instead.

I'm sure the maintainer of the DuckyScriptInterpreter library will be happy to accept a pull request with a fix

> The Combination of STR HOME doesnt work.

What is the STR key??

> Can it be that the delay times are not as long as with original rubber ducky?

DELAY 1000 uses an ESP32 timer to wait for the desired amount of time. It potentially could be 100 longer. Then a whole raft of other tasks are run. So potentially it could be longer.

> File system upload clears my .ds file when uploading

ELI5 what are you doing.

> display file option isnt working at all

What are you expecting to happen when you click that button. Because what the button should do is display an image on the screen of your device.

Besix2, 2025-02-09:
> I'm sure the maintainer of the DuckyScriptInterpreter library will be happy to accept a pull request with a fix<

yeah will look into it. 

> What is the STR key??<

control, strg you name it. But I think this one works. Only home doesnt.

> DELAY 1000 uses an ESP32 timer to wait for the desired amount of time. It potentially could be 100 longer. Then a whole raft of other tasks are run. So potentially it could be longer.

ok thats not really fixable. A bit annyoing because you have to adjust the wait time in ducky scripts.

> ELI5 what are you doing.

When I create a new script.ds file with ducky code and than try to upload it from my phone via the upload feature the file gets wiped and has 0 byte and cant be executed or downloaded(empy files trigger an error when downloading). I know the file is working because when I move it manually on the sd it works. 

p.s. is it possible to refresh the file system after uploading so the files get displayed without restarting(((.(restart button would also be nice :) but thats not really important )))

> What are you expecting to happen when you click that button. Because what the button should do is display an image on the screen of your device.

My mistake didnt realize it is for images.

i-am-shodan, 2025-02-09:
I'm pretty sure STR isn't a key that is defined as a keyword. You need to use CTRL or CONTROL. But if HOME isn't defined correctly you'll need to use RAW_HID command

You can restart the device by sending a RESET command.

What device are you using to upload on? I've tried chrome on windows and safari and haven't seen any problems.

I look at reloading the file cache when an upload occurs.

Besix2, 2025-02-09:
I used a android smartphone with google chrome to upload.
Your hid codes for & are correct I just didnt reallize you have to change the keyboard Layout everytime.
But STR  HOME still doesnt work Im currently trying to figure out why.

Besix2, 2025-02-09:
[image]

In your interpreter: where tf did you define the codes for the system buttons like pause, home, end, scroll down/up ... Everything else is defined...

i-am-shodan, 2025-02-09:
I don't think they're defined, and that's the problem!

Besix2, 2025-02-10:
Created a pull request for your ducky script interpreter. Could you pls test it?

i-am-shodan, 2025-02-12:
I've merged in a slightly modified version of your PR to the DuckyScriptInterpreter library. Will take a few more days for it to make the latest release of UAK.

i-am-shodan, 2025-02-17:
This is now available

closed 2025-02-17 by i-am-shodan