i-am-shodan/USBArmyKnife #73: How to return to original firmware? (im new on this please be kind)

0Proteus117, 2025-02-13:
Ive followed the intrustions with the online website tool and it works good, but i dont find the way to return the original lilygo firmware now and try the other way, in fact i cant even make work the usb mode. (to watch the device in my laptop as a regular usb mass storage)

## Your device

LilyGo T-Dongle S3

# Language (HID attacks only)

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

Sorry in advance if this is a stupid question i tried to search about return original firmware, lilygo install but couldnt find anything about it

rbeede, 2025-02-13:
`python -m esptool --baud 922190 --before default_reset --after hard_reset write_flash 0x0000 yourOriginalFirmwareBackup.bin`

If you need esptool:  `pip install esptool`

0Proteus117, 2025-02-13:
Thank you!! I was just arguing with GPT and found this code

esptool.py --chip esp32s3 --port /dev/tty.usbmodem13101 --baud 460800 write_flash -z 0x0 firmware.bin

And it worked! Thank you so much!

closed 2025-02-14 by i-am-shodan