i-am-shodan/USBArmyKnife #74: Device not always detected as HID device

hegocre, 2025-02-14:
## Your problem

When connecting the device to a computer, either Linux or Windows, it is inconsistently being detected as a Keyboard (TinyUSB) or as a serial device (USB JTAG/serial debug unit). In both cases, the Wi-Fi AP and the web interface start successfully, and I can connect to the webpage and grab the logs (attached later).

If I try to run other payloads stored on the SD card from the web interface (for example, the YouTube part of the Rickroll one), it only works when it is detected as TinyUSB, and not in the other case.

The rest of the settings of the device are the default ones.

As stated before, I've encountered the problem with at least three different computers, using both Windows 10 and Windows 11, as well as Arch Linux.

## Your device

```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)

The Keyboard layout is `win_es`, which I've compiled into the firmware. The keyboard layout on the computer is also ES. There are no errors in the logs regarding this.

## Your DuckyScript

The payload on the device is the following one, which just disables the LED and the screen:
```yaml
TFT_OFF
LED_OFF
```

## Your debug logs

```txt
WifiPw = 'password'
Running!
Setting payload to - '/autorun.ds'
Script finished execution
```

- I confirm I have read the FAQ: ✅ 
- I confirm I am using the latest release: ✅ 

## Additional information
Any more info which could be useful, I will gladly provide it.

Thank you very much for this tool.

i-am-shodan, 2025-02-14:
Definitely odd, I've got a ton of devices and never seen this.

1. Are these machines virtual machines?
2. Are they different physical machines?
3. Can you retest with no autorun.ds file
4. Does the VID/PID of the device change?
5. In device manager are there any keyboards/USB devices in the error state (exclamation mark next to it)
6. Have you changed any of the device settings?
7. When it is detected as a keyboard is there also a serial device?
8. Can you try using the windows tool 'Device Cleanup Tool' and clearing all the entries on your machine and retrying.

hegocre, 2025-02-14:
1. No, they are not virtual machines
2. Yes, at least 4 of them
3. Same thing happens
4. When there is no HID keyboard, I have a serial device (COM3) with id 303a:1001. When the HID is detected, both have id cafe:403f with names Serial Device (COM4) and HID Keybard Device.
5. No
6. No, all settings are default
7. Yes
8. Same result, sometimes the HID device appears, sometimes it doesn't

i-am-shodan, 2025-02-15:
This seems like USB CDC on boot. When the device hits the boot loaded it is starting its own USB stack for serial mode debugging.

It needs to not do that

i-am-shodan, 2025-02-15:
I suspect lilygo is shipping devices with different efuses. It looks like the USB_PHY_SEL eFuse needs to be burnt.

I've had a quick look and it doesn't seem like pio supports setting esp32 efuses in the IDE but there is a python tool espefuse.py that can do it. Maybe the Arduino IDE can do it?

hegocre, 2025-02-15:
So is there anything can I do something to help you confirm this?

i-am-shodan, 2025-02-15:
You could try blowing that eFuse and see if that fixes the issue

hegocre, 2025-02-15:
Should I burn it with a 0 then?

If I understood the documentation correctly, I should run the following command:

```
espefuse.py --port /dev/ttyACM0 burn_efuse USB_PHY_SEL 0
```

Which is returning the following error:
```
A fatal error occurred: New value is not accepted for efuse 'USB_PHY_SEL' (will always burn 0->1), given value=0
```

i-am-shodan, 2025-02-15:
> The ESP32-S3 has a number of eFuses which can store system and user parameters. Each eFuse is a one-bit field which can be programmed to 1 after which it cannot be reverted back to 0.

hegocre, 2025-02-15:
Yep, looks like it is consistently being detected as a HID device. Let me do some verifying on another machine, but seems good now.
Thanks!

i-am-shodan, 2025-02-15:
Can you confirm that setting USB_PHY_SEL to 1 fixed the issue.

Also can you confirm you can still DFU flash the device.

dybbsteppp, 2025-02-15:
If you provide some more instructions I can also test the fix you used to confirm if it's working.

hegocre, 2025-02-15:
> Can you confirm that setting USB_PHY_SEL to 1 fixed the issue.

I can confirm that it fixed the issue.

> Also can you confirm you can still DFU flash the device.

I cannot confirm I can still DFU flash just now, I will be able to do it in a few hours.

i-am-shodan, 2025-02-15:
Great news, it would be great to get the command line you used and I'll update the wiki.

hegocre, 2025-02-15:
> If you provide some more instructions I can also test the fix you used to confirm if it's working.

You need to install the esptool and then run the following command with the device in flashing mode (port may change):

```
espefuse.py --port /dev/ttyACM0 burn_efuse USB_PHY_SEL 1
```

hegocre, 2025-02-16:
Hello. I have checked this, and I was able to program the device with PIO, but with some issues.  The device appears now as `303a:0009 Espressif ESP32-S3` (different name and PID), and I had to do an extra step to be able to flash it.

I had to execute the following command:

```
sudo chmod a+rw /dev/ttyACM0
```
or
```
sudo usermod -aG uudp username
```

Otherwise, I was getting the error `Could not open /dev/ttyACM0, the port doesn't exist`, which wasn't happening before.

Also, the flashing procedure ended with the following error:

```
[...]
Writing at 0x00239e4e... (99 %)
Writing at 0x0023aa5e... (100 %)
Wrote 2274976 bytes (1527532 compressed) at 0x00010000 in 22.8 seconds (effective 799.1 kbit/s)...
Hash of data verified.

Leaving...
WARNING: ESP32-S3 (revision v0.2) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
``` 

Which leads me to believe that the flashing works OK, but it either tries to reset it, and now it can't, or it wasn't trying it before.

Maybe these cannot entirely be considered problems. The first one happened also with arduino, for example, which required you to be root or have the adequate permissions, and the second one could maybe be solved by adding the suggested flag? (I think it is OK to have to reset it manually). Either way, I think it would be good to see if there is a workaround, or document it otherwise.

Thanks!

Talismancer, 2025-02-16:
Hi Hegocre...after this one-way burn of the fuse, will this prevent updates via the Web Installer? Can it see & use the new COM port?

i-am-shodan, 2025-02-16:
If it does, there is undocumented OTA flashing method that I can document. I suspect that it will still work.

Talismancer, 2025-02-16:
That's a pity..my success rate is ~1/4 getting HID. ElegantOTA might be the Jailbreak here.

Talismancer, 2025-02-17:
Can confirm. I just burnt the fuse and it indeed fixed the problem. HID appears every reset. I can update via VScode/PIO but not via the Web Installer just as Hegocre said.

hegocre, 2025-02-17:
Hi @i-am-shodan 

Related to this issue, I've been experimenting with the agent installer script, and I was having some trouble. After investigating a bit, I found that after the agent.img is mounted (USB_MOUNT_DISK_READ_ONLY), the serial and HID devices disappear, and a new serial device with id `303a:1001` appears again, which means that the input commands that should install the agent do not work.

Also, the command WAIT_FOR_USB_STORAGE_ACTIVITY hangs forever at waiting for MSC activity, I temporarily replaced it with simple DELAY, and that's how I found the problem above.

Could this be related to this issue?

Thanks!

i-am-shodan, 2025-02-17:
Worth saying that I checked the efuses on my favourite device which doesn't show any of these issues

From a pio terminal I ran
```
pip install cryptography
pip install ecdsa
pip install bitstring
pip install reedsolo
pio pkg exec --package "platformio/tool-esptoolpy" -- espefuse.py --port COM13 summary
```

USB_PHY_SEL is unset
```
USB_PHY_SEL (BLOCK0)                               Select internal/external PHY for USB OTGand usb_se = False R/W (0b0)
```

hegocre, 2025-02-17:
Maybe is then an issue with another eFuse? Would it be useful if I send you the complete summary?

Hydronus1981, 2025-02-26:
I to have this issue except when it is not occasionally it is never detected as a hid device always shows up ejectable as USB JTAG no HID or ducky scripts will run unless you only want to change the LCD that is all I can get it to do besides wifi and interface there but no payloads will work

for the ESP tool it fails to set port and burn the refuse I am at a loss for my failure on this one

hegocre, 2025-03-10:
@i-am-shodan OK, so I got another device, and this time, it looks like it is working as intended without having to blow the eFuse. Also, the device looks slightly different outside. I don't know if you could confirm, but does your device have a drawing of a TF card over the slot where it is inserted? Also, the last line on the chip is FC00WBJ026, whereas on the non-working device is FC00WBJ025, maybe they are different models?

Cheers!

jd-raymaker, 2025-03-14:
I also had this issue. By blowing the eFuse, it was able to be detected as a USB HID device.
However, I am not able to upload firmware again using PlatformIO.
The upload fails.

```
WARNING: ESP32-S3 (revision v0.2) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
*** [upload] Error 1
==== [FAILED] Took 47.56 seconds ====

Environment         Status    Duration
------------------  --------  ------------
LILYGO-T-Dongle-S3  FAILED    00:00:47.564
```

hegocre, 2025-03-14:
Hey! Yes, it was uploaded fine, it just can reboot it afterwards, but if you look carefully at the logs, you will see the upload got to 100% and was verified!

jd-raymaker, 2025-03-14:
@hegocre you are absolutely correct. It did actually upload successfully!

Thanks for clarifying

i-am-shodan, 2025-03-14:
> @i-am-shodan OK, so I got another device, and this time, it looks like it is working as intended without having to blow the eFuse. Also, the device looks slightly different outside. I don't know if [...]

[image]

[image]

[image]

@hegocre can you share a pic of yours?

i-am-shodan, 2025-03-14:
Anyone know if we can get the model from esptool?

hegocre, 2025-03-14:
> [image]
>
> [image]
>
> [image]
>
> @hegocre can you share a pic of yours?

[image]

This one is from the non-working device 

[image]

And this one is from the working one. Let me know if you need some specific angle

hegocre, 2025-03-14:
> Anyone know if we can get the model from esptool?

The only info I was able to get was the revision of the chip, which in both cases is 0.2:

```
~ esptool.py --port /dev/ttyACM0 -c esp32s3 --after no_reset flash_id

esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: e4:b3:23:f2:bc:10
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4018
Detected flash size: 16MB
Flash type set in eFuse: quad (4 data lines)
Flash voltage set by a strapping pin to 3.3V
Staying in bootloader.
```

Except for the MAC address, the rest of the output is identical on both devices

Onlineraider, 2025-04-14:
> [image]
>
> [image]
>
> [image]
>
> [@hegocre](https://github.com/hegocre) can you share a pic of yours?

My device has the issue with hid

[image]
[image]
[image]
[image]

VladGamer43, 2025-07-16:
Guys i have the issue that after i have blown the efuse i get an error that couldnt Open the Port in VS code