i-am-shodan/USBArmyKnife #76: Feature Request: Scanap File Access

GONZOsint, 2025-02-15:
## Your problem
Access to the generated scanap file, currently an unreadable JSON format, would be valuable.  Making this file accessible would enable a range of possibilities, especially for Marauder related DS scripts, allowing for more sophisticated and dynamic attacks.

## Your device
```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)

What language is your device configured to use? English
 What keyboard layout is your victim machine using? English

## Your DuckyScript
```yaml
ESP32M scanap
```

## Your debug logs
```txt

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-15:
Maybe you could detail some of these possibilities with some fictional ducky script.

You always have the ability to write a custom c++ function which will be able you to parse the file.

GONZOsint, 2025-02-15:
As I understand it, Marauder in its current implementation in usbarmyknife is primarily limited to general attacks on all devices found.

To enable directed attacks, it would be helpful to be able to read the scanap list, especially since Marauder commands already support target specification with -d <dst_mac>, like deauth for specific targets. This would also open the option of the creation of menus in the interface for target selection, or in the web interface.