i-am-shodan/USBArmyKnife #77: Problem: The ESP32-S2 does not display a web interface

EinsPommes, 2025-02-16:
## Problem: The ESP32-S2 does not display a web interface. I successfully flashed my ESP32-S2 with bootloader.bin at 0x1000, partitions.bin at 0x8000, and firmware.bin at 0x10000 without errors; I can connect to the ESP32's Wi-Fi network, and ping 4.3.2.6 works, but accessing http://4.3.2.6 in any browser (Safari, Chrome, Firefox, and Incognito mode) results in a connection failure, with Safari displaying "Page cannot be opened."

## Your device

```txt
ESP32 Key
```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-16:
How have you got to where you are???

From the README where it details ESP32-S2

> Warning They are too underpowered to run the webserver.

Also http://4.3.2.6/ is the URL, it's 4.3.2.1:8080.

EinsPommes, 2025-02-16:
[image]

It still does not work although I am connected to the wifi name iPhone14 and also without VPN

closed 2025-02-16 by i-am-shodan

EinsPommes, 2025-02-16:
ok and how do i reach the website now?

i-am-shodan, 2025-02-16:
You buy an ESP32-S3 based device

EinsPommes, 2025-02-16:
but you wrote that it also works because I have it in the controller design

i-am-shodan, 2025-02-16:
Show me

EinsPommes, 2025-02-16:
[image]

since this esp I have

i-am-shodan, 2025-02-16:
You device works, it is supported, but each device has a different feature profile.

Your device has the same functionality as ESP32 Udisk, it's currently too low on RAM to support a web server and everything else.

It has been a while and the web server is more efficient now. You could try uncommenting '-D NO_WEB' and flashing with that. YMMV.

EinsPommes, 2025-02-16:
where can I do this

i-am-shodan, 2025-02-16:
Check out platformio.ini which specifies what each device supports. Then rebuild and flash

EinsPommes, 2025-02-16:
I have tried it but it does not work, it was supposed to work or

https://www.berrybase.de/waveshare-esp32-s3-1.47-display-dev-board-172-320-262k-color-240mhz-wifi-bluetooth-rgb-led

i-am-shodan, 2025-02-16:
I would advise you get the recommended device