i-am-shodan/USBArmyKnife #78: Agent binary is bigger than partition

Besix2, 2025-02-16:
## Your problem

Everytime I tried to `cp [PATH_TO_AGENT_EXECUTABLE]/* /tmp/mnt/` the agent binary, the command returned no space left.
So I checked with du and and with ls and ls reutrned 500mb and du 501mb while the created partition is only 499mb. (Actually for every iteration of creating agent.img with another lo device the size increases)

[image]

## Your device

```
LilyGo T-Dongle S3

```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-16:
Run

ls -sh /tmp/mnt

Besix2, 2025-02-16:
[image]

i-am-shodan, 2025-02-16:
You need to carefully read the agent instructions. As you seem to be trying to copy agent.img instead of the build artefacts. This is incorrect, quite possibly mad.

What you are meant to be doing is copying the build artefacts from either the GitHub build or a dotnet build.

closed 2025-02-17 by i-am-shodan