i-am-shodan/USBArmyKnife #79: Unable to execute scripts from UI

dybbsteppp, 2025-02-17:
## Your problem

I am unable to run scripts from the UI. I am using the example scripts but clicking execute does nothing from both Status page and from file browser. I can see in the logs the script should be running but nothing happens. When I try to do it from the device using the simple UI it works fine and scrips are executed without issue. 

Usually when the evilAP script is executed the iPhone14 ap disappears and the "free WiFi" ap appears. 

https://github.com/user-attachments/assets/1dd43f68-8b5d-4473-8078-9120cd9514ba

## Your device

```txt

LilyGo T-Dongle S3

```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt

![image](https://github.com/user-attachments/assets/c3a26b7d-a98f-4ed3-837f-ec35158cb704)

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-17:
Can you upgrade to the latest version

i-am-shodan, 2025-02-17:
Sorry I can't reproduce this. I made a script called evil.ds. When onto the web server, executed it and the evilAP started.

The only thing I can think of is you are using a previous build which had this issue.

closed 2025-02-17 by i-am-shodan