i-am-shodan/USBArmyKnife #83: RGB_G Coloring Incorrect?

LaikaSpaceDawg, 2025-02-17:
## Your problem

## Your device

```txt

```

# Language (HID attacks only)

## Your DuckyScript

Simple DS for maintenance/programming access:
```yaml
REM Debugging
REM WIFI_OFF
REM TFT_OFF
REM LED_OFF

LED_OFF

REM Automatically mounts SD card when plugged in
REM Used for maintenance, should not be used in the wild

DEFINE #FILE /lock

IF (FILE_EXISTS() == FALSE) THEN
	LED_G
	REM Mount Entire Contents
	ATTACKMODE STORAGE
	DISPLAY_TEXT 0 70 Storage Mounted
	WAIT_FOR_BUTTON_PRESS
	RESET
ELSE
	REM Do nothing
	LED_R
```

## Your debug logs

```txt

```

- I confirm I have read the FAQ: yes
- I confirm I am using the latest release: yes

## Additional information

While 100,100,100 is correct for HSV -> Green, it appears extremely yellow on the device as pictured: 
(Not the greatest picture, looks more yellow in person)
[image]

Changing to a value like LED 120 100 100 25 works much nicer, and results in a clearer green like this: 

[image]

Just letting you know that it appears that way, and that green does not really result in green so much as yellow.

i-am-shodan, 2025-02-17:
There is already colour correction for some models in src/Devices/LED/FastLED/HardwareLED.cpp. I'd be happy to accept a similar PR for the apa102 driver here src/Devices/LED/APA102/HardwareLED.cpp

i-am-shodan, 2025-07-08:
Fixed in latest release