i-am-shodan/USBArmyKnife #99: Agent not connecting

GrainedPanther, 2025-02-26:
## Agent not connecting

## Your device

```txt

```

# Language (HID attacks only)

## Your DuckyScript

```
Tried uploading the following duckyscript, but couldn't test it. 

DELAY 500
GUI r
DELAY 500
STRING notepad
ENTER
DELAY 1000
STRING Hello, world!
ENTER

```

## Your debug logs

WifiPwd= 'password'
Running!
Could not open file: /autorun.ds
```

- I confirm I have read the FAQ: Yes
- I confirm I am using the latest release: Yes

## Additional information

y-a-r-i-k, 2025-02-26:
Hi, I have the same problem with the same device.
I'm also using the least version of software.

Here's my logs, probably anybody can help me?:

```
WifiPwd = 'password'
Running!
Setting payload to - '/autorun.ds'
FILE_EXISTS() returned 0
Mounting disk /agent.img
Could not load /agent.img
Script error near line 9
```

Probably I didn't understand the docs correctly.

i-am-shodan, 2025-02-26:
> 
> ## Agent not connecting
> 
> 
> 
> ## Your device
> 
> ```txt
> i am using the lilygo T-dongle S3
> ```
> 
> # Language (HID attacks only)
> 
> 
> 
> ## Your DuckyScript
> 
> ```
> 
> 
> ```
> 
> ## Your debug logs
> 
> WifiPwd= 'password'
> Running!
> Could not open file: /autorun.ds
> ```
> 
> - I confirm I have read the FAQ: Yes
> - I confirm I am using the latest release: Yes
> 
> ## Additional information
> 
> Hi, i have purchased the Lilygo T-dongle S3 and am trying to run it as the USBArmyknife. I have flashed it using both ways shown in the github wiki. I flashed it through the web and through Visual studio code. The device gets flashed correctly and the network 'iphone14' works and i can access the webinterface. However it shows that the agent is not connected at the status and the 'connected to value is also blank'. it has maintained this issue each time i boot the device or reflash it. 
> 
> i can run scripts and have full functionality, but the VNC viewer function doesn't work sadly
> 

Sounds like your device is working fine but you didn't install the agent on the host machine.

There is a script in the examples directory to do this. You need to read the instructions and run that script.

i-am-shodan, 2025-02-26:
> Could not load /agent.img

You didn't create the agent.img file as the documentation states. This is the disk that the usb army knife is trying to load and it isn't there.

swinowz, 2025-03-01:
> > Could not load /agent.img
> 
> You didn't create the agent.img file as the documentation states. This is the disk that the usb army knife is trying to load and it isn't there.

I did the agent.img file, the disk is mounted and I can see the files inside. But right now it seems stuck on "Waiting for MSC activity"

i-am-shodan, 2025-03-01:
Please have resolved that by blowing an eFuse on the device

https://github.com/i-am-shodan/USBArmyKnife/issues/74

closed 2025-03-04 by i-am-shodan