i-am-shodan/USBArmyKnife #110: Agent (US) Doesn't work

GrainedPanther, 2025-03-10:
## Your problem

## Your device

```txt
LilyGo T-Dongle S3
transcend premium 32gb simcard
```

# Language (HID attacks only)

## Your DuckyScript

```yaml

```

## Your debug logs

```txt
Web|Rum file/us-autorun.ds
Ducky| setting payload to - 'us-autorun.ds'
Ducky| FILE_EXIST() returned 0
Ducky| Mounting disk /agent.img
error: failed to read from COM port
USB reset 
Disk image mounted 
Waiting for MSC activity
```

- I confirm I have read the FAQ: Yes
- I confirm I am using the latest release: Yes

- My version number is: 1.0.2

## Additional information

i-am-shodan, 2025-03-10:
Your version number is a sha1 hash. Where did you get that number?

You can't run the debug script and the agent at the same time. Stop that script and get the debug from the web hi

GrainedPanther, 2025-03-10:
sorry, i thought the version number was the github release version number.

I attempted to run the agent without the debug script as well. those debug logs are:
WEB debug log:
run file/us_autorun.ds
file_exists() returned 0
mounting disk /agent.img
USB reset 
Disk image mounted 
Waiting for MSC activity

thank you for your quick response!

i-am-shodan, 2025-03-10:
Try doing this https://github.com/i-am-shodan/USBArmyKnife/wiki/Installation#blowing-efuses-to-increase-usb-compatibility-optional

GrainedPanther, 2025-03-10:
have tried the guide but the agent.img still doesn't work:

running the summary gave working output, then ran the burn, this gave an error that the device was not functioning, possibly because of the com port. restarted my laptop and tried again, gave the following output:

pio pkg exec --package "platformio/tool-esptoolpy" -- espefuse.py --port COM3 burn_efuse USB_PHY_SEL 1
Using tool-esptoolpy@1.40501.0 package
espefuse.py v4.5.1
Connecting...
Detecting chip type... ESP32-S3

=== Run "burn_efuse" command ===
The efuses to burn:
  from BLOCK0
     - USB_PHY_SEL

Burning efuses:

    - 'USB_PHY_SEL' (Select internal/external PHY for USB OTGand usb_serial_jtag) 0b1 -> 0b1
        The same value for USB_PHY_SEL is already burned. Do not change the efuse.

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
Nothing to burn, see messages above.

When i retry the agent install example it simply mounts the agent.img and then freezes/ does nothing.

GrainedPanther, 2025-03-10:
command runs, windows button (gui) is pressed. 

- Yes, new keyboard gets added when usb is connected and removed when usb is disconnected
- yes, the start menu open
- no erros in log, script finished execution.

i-am-shodan, 2025-03-10:
I would make sure that efuse is definitely blown because what is happening is that the USB stack is being reset to ensure it is loaded as mass storage but it seems the keyboard component isn't usable anymore. Other people have said the efuse was the fix.

i-am-shodan, 2025-03-12:
@LaertesKrabs do you mean replace the wait for usb storage activity or in addition?

LaertesKrabs, 2025-03-13:
> [@LaertesKrabs](https://github.com/LaertesKrabs) do you mean replace the wait for usb storage activity or in addition?

I added a 5 second delay after the wait for usb storage activity, so the run window opens after the file explorer window, if autoplay is active.

i-am-shodan, 2025-03-15:
Good idea, I've incorporated those changes

closed 2025-03-15 by i-am-shodan