i-am-shodan/USBArmyKnife #112: Storage functionality doesn't work on Waveshare ESP32-S3 LCD

mrapxs, 2025-03-12:
## Your problem

## Your device

```txt

```

# Language (HID attacks only)

## Your DuckyScript

```yaml
DELAY 3000
ATTACKMODE STORAGE
```

## Your debug logs

```txt

```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

- My version number is: Latest (shows up as a hash for some reason)

## Additional information

i-am-shodan, 2025-03-12:
Please use this example https://github.com/i-am-shodan/USBArmyKnife/blob/master/examples/mount_whole_disk_and_run_command/autorun.ds

mrapxs, 2025-03-12:
> Please use this example https://github.com/i-am-shodan/USBArmyKnife/blob/master/examples/mount_whole_disk_and_run_command/autorun.ds

Same behavior, drive mounts but won't load any further, and the GUI r + powershell inputs aren't detected from the device after mounting

i-am-shodan, 2025-03-16:
This should be fixed now in the latest build

closed 2025-03-16 by i-am-shodan

mrapxs, 2025-03-20:
> This should be fixed now in the latest build

There is no release build for this, and there’s no documentation on how to build

i-am-shodan, 2025-03-20:
Check the wiki. You need to pull the latest build from the github action