i-am-shodan/USBArmyKnife #114: Bricked T-Display 3 By Burning USB Chip

Evilander, 2025-03-12:
Burnt USB chip - bricked T-Display Dongle. 

I am unable to flash it now and esptool returns an error when trying to get the device out of download mode.

i-am-shodan, 2025-03-12:
Did you flash usb army knife to it? Does it run? If so you can use the OTA mode to update it

i-am-shodan, 2025-03-12:
Also the codebase has never been tested on any display YMMV

Evilander, 2025-03-13:
> Did you flash usb army knife to it? Does it run? If so you can use the OTA mode to update it

I erased the flash. Blew the efuse. Now I can't do anything with it. When I try to reflash it - no matter what - it either stops at 2 % (online esp) or finishes but can't get out of download mode in VS code - but doesn't function.

i-am-shodan, 2025-03-13:
Erasing was not a great idea. I've updated the wiki to dissuade other people from doing the same

closed 2025-03-13 by i-am-shodan