i-am-shodan/USBArmyKnife #115: Request of example for change the keyboard layout in Wiki

FoxLy1312, 2025-03-13:
Is it just me or could someone else need more information in wiki how to change the Keyboard layout? 

 „ Key presses use the US keyboard layout by default. These can be changed with the command KEYBOARD_LAYOUT.“  

 need a example somehow, I’m keep scrolling through that documentation how and where to paste that and wich options are available .

Idk how to change that label to Question/ Documentation

i-am-shodan, 2025-03-13:
Create a file called autorun.ds on the root of your SD card. Add the following:

```
KEYBOARD_LAYOUT win_en-GB
GUI r
DELAY 1000
STRING im now writing in my language
```

Happy to accept a PR with an example

FoxLy1312, 2025-03-14:
GB layout worked fine, than i tryed German layout and it´s not responding 

`KEYBOARD_LAYOUT win_de-DE
GUI r
DELAY 1000
STRING im now writing in my language`

what am i missing?

i-am-shodan, 2025-03-14:
You need to compile other layouts into the binary. Check out platformio.ini.

FoxLy1312, 2025-03-14:
thanks, appreciate