i-am-shodan/USBArmyKnife #116: [QUESTION] Does this error messages in VS Code caused by missing files matter?

byFynn, 2025-03-15:
The error appears in the **c_cpp_properties.json file** located in the **.vscode folder**.

It says as follows:

**Cannot find: "C/Users/user/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_system/port/soc"** _[Ln 66, Col 17]_

the exact **same error** also appears in _[Ln 375, Col 21]_.

I already tried reinstalling the PlatformIO VS Code extension and also the Framework but the error sadly still persists and I don't know what else to do so I just wanted to know if the error is important or not to safe time on that one.

Thank you for your time and expense!

i-am-shodan, 2025-03-15:
Does the code build successfully?

byFynn, 2025-03-15:
Yes it builds successfully but in the process there are a lot of yellow lines I guess errors but still says its fine at the end.

i-am-shodan, 2025-03-15:
Yellow is warning, if it builds and flashes don't worry about it

closed 2025-03-15 by i-am-shodan

byFynn, 2025-03-15:
TY!