i-am-shodan/USBArmyKnife #118: wifi login issue with esp32-s3

SirCryptic, 2025-03-16:
flashed the latest version from here https://github.com/i-am-shodan/USBArmyKnife/actions/runs/13876968575 and it would consistently tell me the password is wrong when trying to log in to the wifi to access the web interface on device ( esp32-s3 ) , however if i use a version before this it works fine.

i-am-shodan, 2025-03-16:
Well that's unusual, the are no code changes in that version. Only an update to an example file.

The WiFi code is handled by the Arduino framework and then the ESP libraries. Both of those libraries are pinned.

Will see if anyone else reports this issue

i-am-shodan, 2025-03-16:
I tried flashing the latest version to my device. Can log in fine.

closed 2025-03-16 by i-am-shodan