i-am-shodan/USBArmyKnife #133: autorun.ds not working

neorai, 2025-04-09:
## Your problem
autorun.ds not working

## Your device

```txt
LilyGo T-Dongle S3
```

# Language (HID attacks only)
Device language Spanish
keyboard language Spanish

## Your DuckyScript
```yaml
GUI r
DELAY 1000
STRING https://www.youtube.com/watch?v=xvFZjo5PgG0
DELAY 1000
ENTER

REM Now blast the lyrics over WiFI
ESP32M attack -t rickroll

VAR $INDEX = 1
WHILE (TRUE)
    IF ($INDEX == 1)
        DISPLAY_PNG /1.png
    ELSE IF ($INDEX == 2)
        DISPLAY_PNG /2.png
    ELSE IF ($INDEX == 3)
        DISPLAY_PNG /3.png
    END_IF

    $INDEX = ( $INDEX + 1 )
    IF ($INDEX == 4)
        $INDEX = 1
    END_IF
    
    DELAY 500
END_WHILE
```

## Your debug logs
Error: Unable to open COM port COM5

[image]

[image]

- I confirm I have read the FAQ: yes
- I confirm I am using the latest release: yes
- My version number is:

## Additional information

i-am-shodan, 2025-04-10:
Connect to your device over the WiFi interface and post the debug logs

neorai, 2025-04-10:
[image]

i-am-shodan, 2025-04-10:
Did you blow the eFuse?

neorai, 2025-04-10:
> Did you blow the eFuse?

I've tried but it won't let me, it gives the error that it can't open the COM5 port

i-am-shodan, 2025-04-10:
Try putting your device into bootloader mode

neorai, 2025-04-10:
> Try putting your device into bootloader mode

How to do it?

i-am-shodan, 2025-04-10:
You didn't follow the docs or the example. Compiling it in is not enough. You need to tell the script to use that language.

Look at the example script

neorai, 2025-04-10:
> You didn't follow the docs or the example. Compiling it in is not enough. You need to tell the script to use that language.
> 
> Look at the example script

It's working now. In my defense, it doesn't say anywhere to put it in bootloader mode before blowing efuses. Also, is it normal for it to generate these errors in the last step?

The language thing should also be explained in the wiki.

Thanks for your work.

closed 2025-04-10 by i-am-shodan