i-am-shodan/USBArmyKnife #144: Issue with Spanish Layout Firmware on VMware for testing (MSC Not Working)

x224x, 2025-04-18:
## Your problem

Hi I am Shodan,

I'm encountering an issue with the Spanish layout firmware for the LilyGO ESP32-S3 device (with screen) when used inside a VMware virtual machine.

When I connect the device, VMware shows that it's attached to the host (my real PC), even though it should be connected to the virtual machine. The green dot appears, which normally means it's correctly connected to the VM, but the behavior doesn't reflect that. If I try to manually disconnect it from the host, VMware throws an "Unknown error."

I tested the same setup using the original firmware with the English layout, and everything works perfectly—MSC is detected, and agent.img autorun works both on the host and inside the VM.

However, when using the compiled Spanish version of the firmware, MSC does work correctly on the host (autorun runs fine), but not inside the VM. In the logs from the ESP32 device, it keeps saying "Waiting for MSC activity", and the drive never shows up in the VM.

It seems like there's an issue with how the Spanish layout firmware handles USB storage (MSC) in virtual environments.

Thanks in advance for your help!

Best regards,
x224

[image]

[image]

[image]

[image]

## Your device

```txt

```

# Language (HID attacks only)

-D LOCALE_win_es / Machine keyboard es_Spanish

## Your DuckyScript

```yaml

REM Establecer el layout del teclado como español (win_es)
DEFINE KEYBOARD_LAYOUT win_es

REM Para ejecutar esto, primero necesitas crear una imagen de disco que contenga el agente
REM Sigue las instrucciones en tools/README.md
REM Asegúrate de que la imagen se llame agent.img y que esté en la tarjeta SD

DEFINE #FILE /agentInstalled

REM Solo instalamos el agente si es la primera vez que se conecta a esta máquina
IF (FILE_EXISTS() == FALSE) THEN
  USB_MOUNT_DISK_READ_ONLY /agent.img

  WAIT_FOR_USB_STORAGE_ACTIVITY
  DELAY 5000  REM Esperamos a que Windows detecte la unidad

  REM Abrimos el cuadro de ejecutar
  GUI R
  DELAY 1000

  REM Ejecutamos el comando CMD para buscar y lanzar el in1.bat si existe
  STRING cmd /c @echo off ^& for %d in (D E F G H) do if exist %d:\in1.bat %d:\in1.bat #_VID_ #_PID_
  ENTER

  DELAY 1000

  REM Esperamos a que el script copie archivos y cree tareas
  WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP
  DELAY 1000

  REM Creamos el archivo para que no vuelva a instalar
  CREATE_FILE()

  REM Reiniciamos el dispositivo
  RESET
END_IF

REM Esperamos a que el agente se conecte (por WiFi, típicamente)
WHILE (AGENT_CONNECTED() == FALSE)
  DELAY 2000
END_WHILE

REM Ejecutamos un comando para comprobar la conexión
AGENT_RUN dir

```

## Your debug logs

```txt
WifiPwd = 'password'
Running!
Version: 1de5cf2356abf5723eb0463a9363038e006946fc
Setting payload to - '/autorun.ds'
FILE_EXISTS() returned 0
Mounting disk /agent.img
Disk image mounted
Waiting for MSC activity
```

- I confirm I have read the FAQ:
- I confirm I am using the latest release:

- My version number is: 1de5cf2356abf5723eb0463a9363038e006946fc

## Additional information

i-am-shodan, 2025-04-18:
Have you blown the eFuse?

x224x, 2025-04-19:
No, I haven't blown the eFuse yet.
I ran espefuse summary and all the security-related fuses like SECURE_BOOT_EN, SPI_BOOT_CRYPT_CNT, and DIS_DOWNLOAD_MODE are still unset. Everything is still writable and no fuses have been permanently burned.

Also, the current firmware has the Spanish keyboard layout (win_es) already configured.
Interestingly, it works fine on my host machine, but not inside the virtual machine I use for testing.
However, when switching to the English layout, it works on both systems — so I assume that if an eFuse had been blown, it wouldn’t work on the host either, nor with the English layout.

Here's the output of espefuse summary for reference:
## espefuse.py output

```txt
PS C:\Users\999> python3 C:\Users\999\espefuse_tool.py --port COM3 --chip esp32s3 summary
espefuse.py v4.8.1
Connecting...

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
K_RTC_LDO (BLOCK1)                                 BLOCK1 K_RTC_LDO                                   = -8 R/W (0b1000010)
K_DIG_LDO (BLOCK1)                                 BLOCK1 K_DIG_LDO                                   = -40 R/W (0b1001010)
V_RTC_DBIAS20 (BLOCK1)                             BLOCK1 voltage of rtc dbias20                      = -8 R/W (0x82)
V_DIG_DBIAS20 (BLOCK1)                             BLOCK1 voltage of digital dbias20                  = -64 R/W (0x90)
DIG_DBIAS_HVT (BLOCK1)                             BLOCK1 digital dbias when hvt                      = -36 R/W (0b11001)
ADC2_CAL_VOL_ATTEN3 (BLOCK1)                       ADC2 calibration voltage at atten3                 = 112 R/W (0b011100)
TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -7.1000000000000005 R/W (0b101000111)
OCODE (BLOCK2)                                     ADC OCode                                          = 90 R/W (0x5a)
ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = -44 R/W (0x8b)
ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = 0 R/W (0b100000)
ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = 92 R/W (0b010111)
ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = 112 R/W (0b011100)
ADC2_INIT_CODE_ATTEN0 (BLOCK2)                     ADC2 init code at atten0                           = -68 R/W (0x91)
ADC2_INIT_CODE_ATTEN1 (BLOCK2)                     ADC2 init code at atten1                           = -24 R/W (0b100110)
ADC2_INIT_CODE_ATTEN2 (BLOCK2)                     ADC2 init code at atten2                           = 48 R/W (0b001100)
ADC2_INIT_CODE_ATTEN3 (BLOCK2)                     ADC2 init code at atten3                           = 112 R/W (0b011100)
ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = 404 R/W (0x65)
ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = 396 R/W (0x63)
ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = 364 R/W (0x5b)
ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = 408 R/W (0x66)
ADC2_CAL_VOL_ATTEN0 (BLOCK2)                       ADC2 calibration voltage at atten0                 = 476 R/W (0x77)
ADC2_CAL_VOL_ATTEN1 (BLOCK2)                       ADC2 calibration voltage at atten1                 = 216 R/W (0b0110110)
ADC2_CAL_VOL_ATTEN2 (BLOCK2)                       ADC2 calibration voltage at atten2                 = 0 R/W (0b1000000)

Config fuses:
WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 0 R/W (0b0000000)
DIS_ICACHE (BLOCK0)                                Set this bit to disable Icache                     = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Set this bit to disable Dcache                     = False R/W (0b0)
DIS_TWAI (BLOCK0)                                  Set this bit to disable CAN function               = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disable app cpu                                    = False R/W (0b0)
DIS_DIRECT_BOOT (BLOCK0)                           Disable direct boot mode                           = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Set the default UART boot message output mode      = Enable R/W (0b00)
PIN_POWER_SELECTION (BLOCK0)                       Set default power supply for GPIO33-GPIO37; set wh = VDD3P3_CPU R/W (0b0)
                                                   en SPI flash is initialized
PSRAM_CAP (BLOCK1)                                 PSRAM capacity                                     = None R/W (0b00)
PSRAM_TEMP (BLOCK1)                                PSRAM temperature                                  = None R/W (0b00)
PSRAM_VENDOR (BLOCK1)                              PSRAM vendor                                       = None R/W (0b00)
BLOCK_USR_DATA (BLOCK3)                            User data
[... cut ...]
                                                    eFuse to configure VDD_SPI

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
                                                   ock cycle

Flash voltage (VDD_SPI) determined by GPIO45 on reset (GPIO45=High: VDD_SPI pin is powered from internal 1.8V LDO
GPIO45=Low or NC: VDD_SPI pin is powered directly from VDD3P3_RTC_IO via resistor Rspi. Typically this voltage is 3.3 V).
PS C:\Users\999>
```

i-am-shodan, 2025-04-19:
Read the wiki for which eFuse to blow

In both your test cases you are running the script from autorun.ds right and NOT from another file/via the web UI.

When you look in device manager is the device detected as a keyboard AND a mass storage device? My guess is that it isn't.

i-am-shodan, 2025-04-19:
Also you dufus this isn't the line to enable Spanish 

DEFINE KEYBOARD_LAYOUT win_es

It's

KEYBOARD_LAYOUT win_es

x224x, 2025-04-24:
Hi I Am Shodan,

I just wanted to let you know that burning the USB_PHY_SEL eFuse completely solved the issue I was having in my virtual machine — now the device works flawlessly both in virtual and physical environments. Thanks so much for the help and excellent documentation!

I also wanted to ask if it's okay to share your project on LinkedIn. I’m preparing a post aimed at raising awareness about the risks of connecting unknown USB devices, especially for non-technical users in work environments and for family members who aren't familiar with cybersecurity — showing them how an attacker can gain control of a PC with something as simple as a USB stick.

I'd also love to mention your project as a fantastic resource for people just getting started in cybersecurity, and help spread it in the Spanish-speaking community (Spain and LATAM).

If you have a LinkedIn profile, I’d be happy to tag or mention you in the post!

Thanks again for creating such an amazing tool — and congrats on all the work behind it!

Best regards,
x224

closed 2025-04-30 by i-am-shodan