i-am-shodan/USBArmyKnife #147: When I try to run any .ds script I get the message that it could not be executed.

Adesttgrqw21, 2025-04-20:
## Your problem
 When I try to run any .ds script I get the message that it could not be executed. 

## Your device
Lilygo T-Dongle S3
```txt

```

# Language (HID attacks only)
 I use the test scripts from github and none of them work.

## Your DuckyScript

```yaml

```

## Your debug logs
2:28 O
4.3.2.1:8080/index.ht
File Browser VNC Logs
Could not open file: /Evilapp.ds
Script finished execution
.ill 53%
+
13
Help
Run file/Evilapp.ds
Setting payload to - '/Evilapp.ds'
Could not open file: /Evilapp.ds
Script finished execution
Run file/Evilapp.ds
Setting payload to - '/Evilapp.ds'
Could not open file: /Evilapp.ds
Script finished execution
Run file/Evilapp.ds
Setting payload to - '/Evilapp.ds'
Could not open file: /Evilapp.ds
Script finished execution
Run file/Evilapp.ds
Setting payload to - '/Evilapp.ds'
Could not open file: /Evilapp.ds
Script finished execution
Run file/Evilapp.ds
Setting payload to - '/Evilapp.ds'
Could not open file: /Evilapp.ds
Script finished execution 

( I have tried many scripts I have put them in folders, in the root but nothing has worked.)
```txt

```

- I confirm I have read the FAQ: yes
- I confirm I am using the latest release: yes

- My version number is:Device now running
USB MODE: Serial
USB CLASS: HID
Version: 1de5cf2356abf5723
eb0463a9363038e006946fc

## Additional information

I also try re installing firmware 3 times, i try the bad usb attacks on my windows pc the mareuder attacks worked but the bad usb attacks did not (when executing the script) otherwise I left all the settings the same.

[image]

i-am-shodan, 2025-04-20:
Show me a screenshot of the file browser tab.

What happens if you select a script file and press execute?

Adesttgrqw21, 2025-04-21:
https://streamable.com/sga17z I tried with a new card this time in fat 32 and smaller capacity, it still does not send any script to the pc, but tomorrow I will try it in more computers.

i-am-shodan, 2025-04-21:
My guess from your video is that a script is already running. If you look at the status window it says Running.

Try again without an autorun.ds file

Adesttgrqw21, 2025-04-21:
i just reinstalled the latest firmware, and i reset the memory again and i put the rick roll script but it seems that it still does not work if there is something i can do or any recommendation (it says setting payload but i have been waiting 5 minutes so if it does something since it has not changed the lsd picture nor has written anything on the computer) also tomorrow i will try in several different computers 

[image] 

[image]

Adesttgrqw21, 2025-04-21:
[image]

Adesttgrqw21, 2025-04-21:
if I need to activate any special feature in settings let me know, I haven't touched anything in there

Adesttgrqw21, 2025-04-21:
i can record pc too if needed

i-am-shodan, 2025-04-21:
Can you run this script for me and post the log output

```
LOG Script start2
DELAY 1000
LOG delay 1 complete
DELAY 1000
LOG delay 2 complete

REM Now blast the lyrics over WiFI
ESP32M attack -t rickroll
LOG ESP32M complete

VAR $INDEX = 1
WHILE (TRUE)
    LOG while loop
    IF ($INDEX == 1)
        DISPLAY_PNG /1.png
    ELSE IF ($INDEX == 2)
        DISPLAY_PNG /2.png
    ELSE IF ($INDEX == 3)
        DISPLAY_PNG /3.png
    END_IF

    $INDEX = ( $INDEX + 1 )
    IF ($INDEX == 4)
        $INDEX = 1
    END_IF
    
    DELAY 500
END_WHILE

LOG script complete
```

i-am-shodan, 2025-04-21:
Also have you compiled any extra languages in?

Adesttgrqw21, 2025-04-21:
https://streamable.com/7kbjlf video 1 and video 2 https://streamable.com/b7ws2w btw rn thelilygo is turning blue led and re starting in like a period of 7 secs. Nop i havent compiled any extra languages in

Adesttgrqw21, 2025-04-21:
maybe i should efuse?

i-am-shodan, 2025-04-21:
All I want is the logs, not videos

Adesttgrqw21, 2025-04-21:
alr i found the solution dw ty for help and support

i-am-shodan, 2025-04-21:
What was the problem / solution?

i-am-shodan, 2025-04-21:
Dude, let us in on the secret

Adesttgrqw21, 2025-04-21:
sd card was corrupted, never buy a pack of 3 sd cards from aliexprex