i-am-shodan/USBArmyKnife #157: [QUESTION] How to turn LED(s) completely off? (Lilygo TDongle S3)

byFynn, 2025-05-01:
Where in your project for Lilygo TDongle S3 can the LED(s) be fully turned off?

I searched everything (every file) and couldn't find a define option for that purpose.
I also tried turning the LEDs of with changing some files and with the autoruns.ds file on the TDongle, but that sadly somehow didn´t work (The .ds script worked but the autoruns file could not be opened/accessed {seen in logs of the device}, and that's why it did not start on boot).
I also checked the file permissions and reflashed the Dongle, but also that did not help.

I myself don't need the LED(s), at least after boot, and they are pretty ugly on the Dongle (mix between light yellow and green). That's why I am asking you for the/a solution to my problem.

Thank you for your response! 

PS: Great project keep up the good work!

BenjaminGade, 2025-05-01:
LED_OFF and LED_ON is the commands you need, it's in the wiki under commands. Add it to the start of your autorun.ds file to shut off the led when the device is plugged in. If you need the screen to be off then use TFT_OFF and TFT_ON to turn it on again.

BenjaminGade, 2025-05-01:
The wiki can be found here: https://github.com/i-am-shodan/USBArmyKnife/wiki

byFynn, 2025-05-01:
TYSM Benjamin!

closed 2025-05-01 by byFynn