i-am-shodan/USBArmyKnife #166: Simple UI duckyscript does not work at all

rocord01, 2025-06-01:
## Your problem

The "Simple UI" duckyscript is not working whatsoever, seemingly hanging on "LOAD_DS_FILES_FROM_SD()"

## Your device

```txt
Device: LilyGo T-Dongle S3
SD: 32GB Samsung EVO Select
```

# Language (HID attacks only)

N/A

## Your DuckyScript

```yaml
VAR $FILE_INDEX = 0

LOAD_DS_FILES_FROM_SD()

WHILE ( FILE_INDEX_VALID() == TRUE )
    DISPLAY_PNG /select.png
    DISPLAY_TEXT 0 35 #_FILE_INDEX_FILE_NAME_

    DISPLAY_TEXT 0 55 Click) Next file
    DISPLAY_TEXT 0 65 Hold ) Run

    WAIT_FOR_BUTTON_PRESS

    IF (BUTTON_SHORT_PRESS())
        $FILE_INDEX = ( $FILE_INDEX + 1 )
    ELSE IF (BUTTON_LONG_PRESS())
        DISPLAY_PNG /select.png
        DISPLAY_TEXT 0 35 Run #_FILE_INDEX_FILE_NAME_
        DISPLAY_TEXT 0 45 Are you sure?

        DISPLAY_TEXT 0 55 Click) Back
        DISPLAY_TEXT 0 65 Hold ) Run
        WAIT_FOR_BUTTON_PRESS

        IF (BUTTON_LONG_PRESS() == 1)
            DISPLAY_CLEAR
            RUN_PAYLOAD #_FILE_INDEX_FILE_NAME_
        ELSE
            REM DO NOTHING
        END_IF
    END_IF
END_WHILE

RESTART_PAYLOAD
```

## Your debug logs

```txt
WifiPwd = 'password'
Running!
Version: c18a211f8ed36b41f8139dae441c554a97827777
Setting payload to - '/autorun.ds'
Script finished execution
```

- I confirm I have read the FAQ: yes
- I confirm I am using the latest release: yes

- My version number is: c18a211f8ed36b41f8139dae441c554a97827777

## Additional information

SD Dump: [03DA-FFC6.zip](https://github.com/user-attachments/files/20542806/03DA-FFC6.zip)

rocord01, 2025-06-01:
I have tested the device with:
```yaml
DISPLAY_CLEAR
DISPLAY_TEXT 0 0 Press the button
LED_R
WAIT_FOR_BUTTON_PRESS
LED_G
WAIT_FOR_BUTTON_PRESS
LED_B
WAIT_FOR_BUTTON_PRESS
```
and the script seems to work just fine.

i-am-shodan, 2025-06-27:
Are you still having this issue? Before I start debugging can I ask if files appear in the file browser tab in the web UI

rocord01, 2025-06-28:
Yes, I am still having this issue, and the files do appear on the browser UI.

i-am-shodan, 2025-07-08:
Fixed in latest release

closed 2025-07-08 by i-am-shodan