i-am-shodan/USBArmyKnife #168: FEATURE Add support for EvilCrowCable Wind cables

i-am-shodan, 2025-06-17:
The joelsernamoreno/EvilCrowCable-Wind#21 project allows you to buy ESP32-S3 equipped USB cables. These would make perfect USB Army Knife devices with little change.

joelsernamoreno, 2025-06-17:
ChatGPT
Tú dijiste:
Esto está bien escrito en inglés?

Hello

I would love to see a USBArmy firmware for my Evil Crow Cable Wind

In order to do this, I think you have to have these considerations:

- Evil Crow Cable Wind has an ESP32-S3 with 8MB of flash:

Flash Size - "8MB (64Mb)"
Partition Scheme - "8M with SPIFFS (3MB APP/1.5MB SPIFFS)"

- Evil Crow Cable Wind can only operate in STA mode. Being a very small PCB, the ESP32-S3 chip gets very hot when in AP mode and can break when reaching high temperatures.

Can your firmware meet this?
ChatGPT Plus
Tu texto está bastante claro, pero aquí tienes una versión corregida y más natural en inglés:

Hello,

I would love to see USBArmy firmware available for my Evil Crow Cable Wind.

To make this possible, I believe the following considerations need to be taken into account:

The Evil Crow Cable Wind uses an ESP32-S3 with 8MB of flash memory:
Flash Size: "8MB (64Mb)"
Partition Scheme: "8M with SPIFFS (3MB APP / 1.5MB SPIFFS)"

The Evil Crow Cable Wind can only operate in STA mode. Due to the small size of the PCB, the ESP32-S3 chip overheats in AP mode, which can cause permanent damage if high temperatures are reached. The firmware for Evil Crow Cable Wind should only allow STA mode

Would your firmware be compatible with this setup?

Thanks

i-am-shodan, 2025-06-17:
Thank you for taking the time to reply. The details are really useful and I've already bought two cables to try with.

The USB Army Knife supports a large range of devices (ESP32 S2/S3 and RP2040). Each of these support different hardware so it should be pretty easy to support the Evil Crow Cable Wind.

I may create a new partition scheme to try and get a slightly larger SPIFFS partition. Disabling OTA for instance saves quite a bit.

Thank you for mentioning the thermal issues. I will do some testing. In the worst case scenario I can disable AP mode but I will also try reducing the TX power. I can also just wait if the ESP32 temperature sensor is reading too high.

Does the ESP32 have any PSRAM? I suspect not given the size.

i-am-shodan, 2025-06-18:
Good to know. I don't currently enable PSRAM for the majority of use cases as I've noticed instability with the HTTP server in use.

i-am-shodan, 2025-06-27:
@joelsernamoreno the max working temperature of the ESP32-S3-PICO-1-N8R2 is 85C. Do you have a good working temperature for the wind cable?

I'm adding overheating protection so at 60C WiFi will go off until the unit cools down. Is this too high?

i-am-shodan, 2025-07-01:
Did some investigations today on my USB-A to USB-C cable.

* Internal temperatures can easily and quickly (2-3m) hit >100C in WiFi AP mode even with the minimum power usage.
* With WiFi off I get an average temperature of 63C
* In AP mode with low power and WiFi sleep enabled I see temperatures of around 75C

I've added code to immediately disable WiFi if it gets too hot. It then gets enabled when it cools down. Seems to work.

I will ensure the docs say it really needs to run in station mode.

joelsernamoreno, 2025-07-01:
For example, the ESP32-S3-ZERO board also gets too hot in AP mode

joelsernamoreno, 2025-07-01:
Believe me it is difficult to do so many things inside a USB-A connector, in the USB-C version it is even more complicated, this is very small 🥲🥲

i-am-shodan, 2025-07-01:
Yeah completely agree. I was very surprised you got an ESP32 to fit!

The only thing I can suggest is to make the enclosure metal.

i-am-shodan, 2025-07-08:
Added in latest release